mikakaraila/node-red-contrib-opcua #297: user login problem

dvdvelas, 2021-05-18:
I try to use credential for the OPC-UA Server but y cant connect with the Client "Client node error on:  error: [" OPC UA connection error: BadUserAccessDenied (0x801f0000)"]"
[image]

Server Node
[image]

Client Node
[image]

mikakaraila, 2021-05-18:
And you have created User JohnSmith with password to Prosys UA Simulation server?

dvdvelas, 2021-05-18:
I use de default users in "user.json"
[image]

mikakaraila, 2021-05-18:
It could be that the latest node-opcua will accept only WellKnownRoles. Breaking change in the API.
I will check this later....

mikakaraila, 2021-05-18:
I just tested and it is working. Can you check you environment, is there another server on same port?

[image]

mikakaraila, 2021-05-18:
I tested with these settings:

[image]

dvdvelas, 2021-05-18:
yes is the only server in this port, i try your configuration but I have the same problem

with this configuration i can connect without user:
[image]

but if i use a user i have this error:
[image]

mikakaraila, 2021-05-18:
Hmm, can you see any error messages on node-red console?

dvdvelas, 2021-05-19:
Yes i see this error "Client node error on:  error: [" OPC UA connection error: BadUserAccessDenied (0x801f0000)"]"

[image]

And on UaExpert

[image]

I tested with and without all options

dvdvelas, 2021-05-19:
Do I have to mark any option for the server to use the file "user.json"?

mikakaraila, 2021-05-19:
File is actually users.json:

        verbose_log("Loading default users from file: users.json");
        // From the node-red-contrib-opcua folder or input as opcuaCommand
        if (fs.existsSync("users.json")) {
            users = JSON.parse(fs.readFileSync("users.json"));
            verbose_log("Users: " + JSON.stringify(users));
            setUsers(users);
        }

Please run from command line to see possible errors with command like: node-red -v test.json

mikakaraila, 2021-05-19:
I added error message if file is not found. This could be the reason your environment is not working.

I have copied users.json to same folder with flow files.

dvdvelas, 2021-05-19:
i add the line
[image]

and i see this 
[image]

i have the  file "user.json" in this directorys
 
C:\Users\INGAUT_DVL\.node-red           <--- this is the flows folder
C:\Users\INGAUT_DVL\.node-red\node_modules\node-red-contrib-opcua\opcua
C:\Users\INGAUT_DVL\.node-red\node_modules\node-red-contrib-opcua

mikakaraila, 2021-05-19:
Filename is users.json

dvdvelas, 2021-05-20:
I was wrong to write the comment, this is the file in the flows folder
[image]

i fix the problem 

[image]

change "fs.existsSync("users.json")" for "fs.existsSync("./.node-red/users.json")" and "fs.readFileSync("users.json") for" "fs.readFileSync("./.node-red/users.json")" .This works for me.

now i can read the file "users.json" in flows folder
[image]

[image]

Thanks a lot for help :)

closed 2021-05-20 by dvdvelas

mikakaraila, 2021-05-20:
OK, I have to check how to make it official. I mean there are different installations.

I will do change to next version.

But you can also inject users as payload to server.

dvdvelas, 2021-05-20:
yes but if i insert a user i lose the others users , i prefer a file that i can modify and store

dvdvelas, 2021-05-20:
one cuestion, in this part of code
[image]

the last two would not be "ConfigureAdmin"not  "ConfigureAmin"?

mikakaraila, 2021-05-20:
Yes, my typo error. I will fix it also.

mikakaraila, 2021-05-20:
Update to new v0.2.221 it contains now fixed code and parameter for users.json file + better error messages.

dvdvelas, 2021-05-31:
Hello i tested de new version and works perfectly ,  i have new cuestion ,what is the function off the tick "Auto Accept Unknown Certificates " i prove with and without  this opcion  and always  can connect

dvdvelas, 2021-05-31:
and thanks a lot for improviments

mikakaraila, 2021-05-31:
If server has once got certificate then it is not anymore "unknown".