mikakaraila/node-red-contrib-opcua #299: Extension Object can't be created inside OPC-UA Server

gteti, 2021-05-21:
I can't execute an "addVariable" command for an ExtensionObject.

The output show this error when executed:

> OPC-UA Server
> msg : error
> "TypeError: Cannot read property 'toString' of null"

Flow: 
[{"id":"b2238d92.fd329","type":"inject","z":"f6f2187d.f17ca8","name":"New variable siglElem","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":"","topic":"ns=6;s=siglElem;datatype=ExtensionObjectType","payload":"{ \"opcuaCommand\": \"addVariable\" }","payloadType":"json","x":220,"y":1520,"wires":[["f810ef29.b0ede"]]},{"id":"f810ef29.b0ede","type":"OpcUa-Server","z":"f6f2187d.f17ca8","port":"53880","name":"OPC-UA Server","endpoint":"UA/SimpleNodeRedServer","autoAcceptUnknownCertificate":true,"registerToDiscovery":false,"constructDefaultAddressSpace":false,"allowAnonymous":true,"endpointNone":true,"endpointSign":false,"endpointSignEncrypt":false,"endpointBasic128Rsa15":false,"endpointBasic256":false,"endpointBasic256Sha256":false,"maxNodesPerBrowse":"","maxNodesPerHistoryReadData":"","maxNodesPerHistoryReadEvents":"","maxNodesPerHistoryUpdateData":"","maxNodesPerRead":"","maxNodesPerWrite":"","maxNodesPerMethodCall":"","maxNodesPerRegisterNodes":"","maxNodesPerNodeManagement":"","maxMonitoredItemsPerCall":"","maxNodesPerHistoryUpdateEvents":"","maxNodesPerTranslateBrowsePathsToNodeIds":"","x":560,"y":1400,"wires":[["ca867ae.353c888"]]},{"id":"ca867ae.353c888","type":"debug","z":"f6f2187d.f17ca8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","statusVal":"","statusType":"auto","x":770,"y":1420,"wires":[]}]

mikakaraila, 2021-05-21:
As I said it's not yet complete.
It needs bit more work...

gteti, 2021-05-21:
How can it be done with the .xml ? If I can implement it, I'll add a PR.

mikakaraila, 2021-05-22:
First practice modeling with some UaModeler like https://github.com/FreeOpcUa/opcua-modeler

Then I have to add support for custom nodeset that will load it.

mikakaraila, 2021-05-23:
OK, I have prepared custom nodeset loading and added ExtensionObject to addVariable, but I can test this on Monday.

[image]

mikakaraila, 2021-05-24:
Still some work to do, but progressing:

[image]

gteti, 2021-05-25:
How is composed the user.json ? I have some issue importing the example file from https://github.com/FreeOpcUa/opcua-modeler on that program instance.

mikakaraila, 2021-05-25:
user.json is for username, password and role.

I am still working with custom.xml that contains type definition. Seems to be some strange things how it is loaded and handled at the server. Needs still some clarification before I can commit code & example.

mikakaraila, 2021-06-03:
New version contains support for custom nodeset xml but I didn´t have time to test instantiate extension object.
Closer but not yet final... high workload at work.

haph-vi, 2021-10-07:
Hi!
I need to create a structure like in the picture. 
I can't find the right settings within opcua-modeler for that.

Within a node-red function block tried it with 
`msg.payload={"opcuaCommand": "addVariable"};
    msg.topic='ns=1;s=Systest_Daten;datatype=ExtensionObject';`
but there are errors:

`"Error: String cannot be coerced to a nodeId : =Systest_Daten;datatype=ExtensionObject"`

Is it possible now and am I doing something wrong or is it not implemented yet?

Thanks!

[image]

mikakaraila, 2023-02-28:
I expect this one can be closed already.

closed 2023-02-28 by mikakaraila