mikakaraila/node-red-contrib-opcua #300: subscription cannot be deleted

tord-v, 2021-05-27:
I have set the action to SUBSCRIBE. (And by feeding the OpcUa-Client node items, the subscription works just fine.)

I have tried the following msg's to delete the subscription. But with no luck:
```
msg={'action':'deletesubscription'}
msg={'payload':{'action':'deletesubscription'}
msg={'action':'deletesubscribtion'}
msg={'payload':{'action':'deletesubscribtion'}
```

mikakaraila, 2021-05-27:
What about unsubscribe ?

tord-v, 2021-05-28:
That works just fine. I actually use it that way anyway (by unsubscribing to each nodeid individually). 

Its just nice to be able to delete the subscription.

Great work with the node btw! This might not be the right forum for this, if I create a secondary opcuaclient-node. Will this one handle its own subsciptions independent of the first node? (using the same endpoint config)

mikakaraila, 2021-05-30:
TEST-NODES contains example:
[image]

[image]

mikakaraila, 2021-06-10:
Tested and works:

[image]

mikakaraila, 2021-06-10:
Client node shows status: terminated

[image]

mikakaraila, 2021-06-10:
Inject node:

[image]

closed 2021-06-10 by mikakaraila