mikakaraila/node-red-contrib-opcua #301: write a Boolean values for multiple node

stesquality, 2021-05-28:
In my case, I can get the msg.payload "a", "b" or "c" (means I have 3 nodes) and I want if my msg.payload is "a" then in OPC-UA client end it will automatically get the value of a is "1" and others nodes value "0". Simarlarly, if I will get mag.payload is "b" then in OPC-UA client end it will automatically get the value of a is "1" and other nodes value "0" and vice versa. Do you have any suggestion,
[image]

mikakaraila, 2021-05-29:
Boolean value is true or false

stesquality, 2021-05-29:
yes or it can be "0","1".

mikakaraila, 2021-05-30:
TEST-NODES contains example:
[image]

[image]

stesquality, 2021-05-31:
Thanks for your guidance, but how the system will work, I mean as I can get a, b, or c as an output. That's means there will be three variables and then the output variable will show a boolean value.

mikakaraila, 2021-06-01:
This way:
[image]

[ABC.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/6575470/ABC.txt)

stesquality, 2021-06-01:
Thanks for you kind suggestion but where I need to attach OPC-UA.

mikakaraila, 2021-06-02:
This way:
[image]

mikakaraila, 2021-06-02:
I found one mistake from Boolean Array converting code it will accept 1 as true and "true" as true, but not true as true.
This was missing, I just added it to the code.

mikakaraila, 2021-06-03:
Here is example and new v0.2.222 works with this.
[ABC.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/6589096/ABC.txt)

stesquality, 2021-06-03:
First of all thanks for your guidance, Actually I need something like as as shown in below picture.
[image]

On the other hand, I have added the file you attached, but i am getting error as mentioned in below picture.
[image]

mikakaraila, 2021-06-03:
Those nodeId variables are NOT Boolean Array they are single Boolean variables.
I suppose you should solve your problem by yourself. It is now very simple basic solution that is needed.
In case of variable A you should write just single Boolean variables to your server and same way with other variables.

closed 2021-06-03 by mikakaraila

stesquality, 2021-06-07:
Actually, I have tried as showed in attachment, but i did not get the required result.
[image]
[flows.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/6610380/flows.txt)

mikakaraila, 2021-06-07:
Steps:
1) Use on client node your robot server endpoint
2) Make 3 item nodes with correct nodeId like ns=4;i=3;datatype=Boolean
3) Set input to switch when you want to set them

Or use one function node with multiple node.send() to set each nodeId (not so visible but fast & easy to do).

mikakaraila, 2021-06-07:
Here is example, it can be reduced a lot but this way it is perhaps more informative.

[ABC.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/6611464/ABC.txt)

stesquality, 2021-06-09:
Thanks again for your kind support, Actually, I tried and client is showing, connected and I am also not getting any error, but it is not changing the variables values in server end. As mentioned in picture the "NamespaceIndex" is 4, so I changed it and I also changed the "Identifier" of each variables(3,4,5). And my server is in TIA Portal V16. Do you have any suggestion?
[image]

mikakaraila, 2021-06-09:
Can you provide your flow or write part of it here?
Or run from command line:
node-red -v ABC.json
Then you can see values written on console

stesquality, 2021-06-09:
[flows (1).txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/6624139/flows.1.txt)
It is the same file what you sent me but as I told you I NamespaceIndex"  and  "Identifier" .

mikakaraila, 2021-06-09:
Hmm, in UaExpert nodeId is ns=4;i=3 they are integer based indexed nodeIds and you have in the flow still ns=4;s=3 string based addresses. It is mistake in your flow.

[image]

stesquality, 2021-06-10:
Thanks, It is working...