mikakaraila/node-red-contrib-opcua #303: I'm on Close error

paulsalibe, 2021-06-01:
I am getting a strange error when I check the status of my node red, not sure if it is from Node-OPCUA but there is also an accompanying error with it from Node-OPCUA. I have attached a screenshot showing what I am seeing.. The strangest part is it that it still reads/writes some of the time but others it does not. 
[image]

mikakaraila, 2021-06-01:
Warning comes from node-opcua. Client node/nodes are writing too fast/multiple times and server cannot process them fast enough. There is warning limit after request queue is over 10.

mikakaraila, 2021-06-18:
Can you attach flow to check if it will send multiple writes to server immediately?
You could perhaps use write multiple action...

mikakaraila, 2021-07-17:
Did you test with write multiple? If no actions, I will close this issue.

closed 2021-07-17 by mikakaraila