mikakaraila/node-red-contrib-opcua #304: Read entire INT array not working

Julianito98, 2021-06-04:
Hello everyone,

I am using the node-red-contrib-opcua package in node-red.
I want to read all values of an INT array in a opcua server.
I can easily read the values one by one with :  msg.topic = "s=2;ns=Mytab[3]" for example. I tried all types of "Type" of values.
But i can't read all values in the same time.

Can someone help me please ?
Thanks a lot, and have a good day !

mikakaraila, 2021-06-04:
Did you specify Type as here:

[image]

mikakaraila, 2021-06-04:
Works with Prosys UA Simulation server:

[image]

[image]

Julianito98, 2021-06-04:
Yes i specified the type like that.

I think i have an issue because my OPCUA PLC Server give me the array as a folder with values : 

[image]

Do you know how i can overcome this?

Thanks a lot!

mikakaraila, 2021-06-04:
Those are single scalars. Look each variable nodeID and you can write them one by one

mikakaraila, 2021-06-09:
Server does not implement real Array variable as it should. It seems to provide normal scalar variables.

closed 2021-06-09 by mikakaraila