mikakaraila/node-red-contrib-opcua #306: Get the NodeId in notification

anxuae, 2021-06-08:
Hi,

Currently, after subscribing/monitoring a variable, 3 attributes can be retrieved from the notification:
```json
payload: object
    messageType: "Variable"
    variableName: "call"
    variableValue: false
```

Is it possible to get the NodeId (ns=x;s=MyVar) from the notification?
It would be really usefull to distinguish 2 different nodes with same variable name.

anxuae, 2021-06-08:
Sorry, in fact I didn't look at the correct place

mikakaraila, 2021-06-09:
I will check & add.

mikakaraila, 2021-06-09:
Added to payload:

[image]

mikakaraila, 2021-06-09:
Available v0.2.223 please test & close

anxuae, 2021-06-10:
Tested ok

closed 2021-06-10 by anxuae