mikakaraila/node-red-contrib-opcua #307: Using 2 nodes with same name and different name space

anxuae, 2021-06-08:
It seems that using 2 nodes with same name but different namespace does not work correctly:
Setting the value of the first node will change the value of the second node.

Here below is my example:

[image]

```json
[{"id":"a257ac9d.07628","type":"tab","label":"Flow 2","disabled":false,"info":""},{"id":"7a3080c8.b6de4","type":"OpcUa-Server","z":"a257ac9d.07628","port":"53801","name":"LocalServer","endpoint":"MyServer","users":"","nodeset":"","autoAcceptUnknownCertificate":true,"registerToDiscovery":true,"constructDefaultAddressSpace":false,"allowAnonymous":true,"endpointNone":true,"endpointSign":false,"endpointSignEncrypt":false,"endpointBasic128Rsa15":false,"endpointBasic256":false,"endpointBasic256Sha256":false,"maxNodesPerBrowse":"","maxNodesPerHistoryReadData":"","maxNodesPerHistoryReadEvents":"","maxNodesPerHistoryUpdateData":"","maxNodesPerRead":"","maxNodesPerWrite":"","maxNodesPerMethodCall":"","maxNodesPerRegisterNodes":"","maxNodesPerNodeManagement":"","maxMonitoredItemsPerCall":"","maxNodesPerHistoryUpdateEvents":"","maxNodesPerTranslateBrowsePathsToNodeIds":"","x":850,"y":240,"wires":[["f35a035f.0714d"]]},{"id":"dc0f027b.9818b8","type":"function","z":"a257ac9d.07628","name":"Create NameSpace 1","func":"msg.payload = { \"opcuaCommand\" : \"registerNamespace\" };\nmsg.topic = \"Namespace 1\";\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":480,"y":180,"wires":[["7a3080c8.b6de4"]]},{"id":"aaa09827.1f2d","type":"function","z":"a257ac9d.07628","name":"Create NameSpace 2","func":"msg.payload = { \"opcuaCommand\" : \"registerNamespace\" };\nmsg.topic = \"Namespace 2\";\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":480,"y":260,"wires":[["7a3080c8.b6de4"]]},{"id":"f665b244.1dcbe","type":"function","z":"a257ac9d.07628","name":"Add MyVariable Variable","func":"msg.payload = {\"opcuaCommand\":\"addVariable\"};\nmsg.topic = \"ns=5;s=MyVariable;datatype=Boolean\";\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":490,"y":220,"wires":[["7a3080c8.b6de4"]]},{"id":"d231534f.afd0e8","type":"function","z":"a257ac9d.07628","name":"Add MyVariable Variable","func":"msg.payload = {\"opcuaCommand\":\"addVariable\"};\nmsg.topic = \"ns=6;s=MyVariable;datatype=Boolean\";\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":490,"y":300,"wires":[["7a3080c8.b6de4"]]},{"id":"f35a035f.0714d","type":"debug","z":"a257ac9d.07628","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":1110,"y":240,"wires":[]},{"id":"c45a16e8.8c185","type":"inject","z":"a257ac9d.07628","name":"","props":[],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":250,"y":180,"wires":[["dc0f027b.9818b8"]]},{"id":"b9d50fa1.f173a8","type":"inject","z":"a257ac9d.07628","name":"","props":[],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":250,"y":220,"wires":[["f665b244.1dcbe"]]},{"id":"71f127b1.2e4938","type":"inject","z":"a257ac9d.07628","name":"","props":[],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":250,"y":260,"wires":[["aaa09827.1f2d"]]},{"id":"898f7c1d.9a27a","type":"inject","z":"a257ac9d.07628","name":"","props":[],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":250,"y":300,"wires":[["d231534f.afd0e8"]]},{"id":"a582a8ec.df5ba8","type":"OpcUa-Item","z":"a257ac9d.07628","item":"ns=5;s=MyVariable","datatype":"Boolean","value":"","name":"write","x":430,"y":380,"wires":[["c04a6c52.d38c38"]]},{"id":"c04a6c52.d38c38","type":"OpcUa-Client","z":"a257ac9d.07628","endpoint":"995ba144.913f08","action":"write","deadbandvalue":"","time":10,"timeUnit":"s","certificate":"n","localfile":"","localkeyfile":"","securitymode":"SignAndEncrypt","securitypolicy":"Basic256","name":"LocalServer WRITE","x":880,"y":380,"wires":[["a2d415a8.729bc"]]},{"id":"a2d415a8.729bc","type":"debug","z":"a257ac9d.07628","name":"Client output","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":1110,"y":380,"wires":[]},{"id":"49bcd50e.efb324","type":"inject","z":"a257ac9d.07628","name":"Set true","props":[{"p":"payload"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"true","payloadType":"str","x":250,"y":380,"wires":[["a582a8ec.df5ba8"]]},{"id":"adf7b3c8.63ace","type":"inject","z":"a257ac9d.07628","name":"Set false","props":[{"p":"payload"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"false","payloadType":"str","x":260,"y":440,"wires":[["a582a8ec.df5ba8"]]},{"id":"995ba144.913f08","type":"OpcUa-Endpoint","endpoint":"opc.tcp://localhost:53801/MyServer/","secpol":"None","secmode":"None","login":false}]
```

mikakaraila, 2021-06-09:
Your another issue and earlier are related both to same problem. As I extended namespace it was not considered fully.
I will test and fix this.

mikakaraila, 2021-06-09:
OK now code uses nsindex as prefix (nsindex + ":" browsename) in variables[] and works:

[image]

mikakaraila, 2021-06-09:
Available v0.2.223 please test & close

anxuae, 2021-06-10:
Tested ok

closed 2021-06-10 by anxuae