mikakaraila/node-red-contrib-opcua #308: Read/write ArrayString in node

anxuae, 2021-06-09:
Hi,

I've try to use an array of string in a OPC-UA node, and try to read / write it.
It seems to work perfectly 😄 , event if it is not described in the doc and not present in the `OpcUa - item` node. Does it a real possibility? Or I'm using a hack?

If it is a real feature, it could be good to add it in the doc and `OpcUa - item` node.

Here is the function node used to read from the client side
```javascript
msg.payload = { "opcuaCommand" : "addVariable" };
msg.topic = `${msg.ns};s=faults;datatype=StringArray[200]`;
return msg;
```

Here is the function node used to read from the client side
```javascript
msg.topic = "ns=7;s=faults";
msg.datatype = "datatype=StringArray";
msg.browseName = "faults";
return msg;
```

mikakaraila, 2021-06-09:
Yes, code itself is generic, but I have missed to add it to Item-node.

mikakaraila, 2021-06-09:
Was not fully working, I fixed couple of things and now it works. I added it to item-node and documentation.

[image]

mikakaraila, 2021-06-09:
Available v0.2.223 please test & close

anxuae, 2021-06-10:
Tested ok

closed 2021-06-10 by anxuae