mikakaraila/node-red-contrib-opcua #309: How to define default value on a node from sever side?

anxuae, 2021-06-09:
Hi,

Is is possible to define a default value when creating a node in the server?

One possibility will be to define at the node creation:

```javascript
msg.payload = { "opcuaCommand" : "addVariable" };
msg.topic = `ns=1;s=call;datatype=Boolean;value=true`;
return msg;
```

Other possibility is to give access to write/read data on the server by adding new commands, moreover those commands could be useful to develop a server and change variables without the used of a client in the server flows. For instance:

```javascript
msg.payload = { "opcuaCommand" : "writeVariable" };
msg.topic = `ns=1;s=call;value=true`;
return msg;
```

```javascript
msg.payload = { "opcuaCommand" : "readVariable" };
msg.topic = `ns=1;s=call`;
return msg;
```

mikakaraila, 2021-06-09:
OK

mikakaraila, 2021-06-09:
Those read/write variables are actually not a good idea. Better to use UaExpert to test server variables.
Otherwise there can be something wrong with real variable set/get functions.

mikakaraila, 2021-06-09:
Seems to work:
[image]

[image]

mikakaraila, 2021-06-09:
Available v0.2.223 please test & close

anxuae, 2021-06-10:
Tested ok

closed 2021-06-10 by anxuae