mikakaraila/node-red-contrib-opcua #310: Reading the performance view data from UAexpert into a csv file

ankita1909, 2021-06-09:
Hello,

I am new to working with OPC UA. In UAexpert i can visualise the performance view for reading and writing of the data. But is it possible to get these values into a csv file. I could not find any local files that are generated when i try to read/write data and the result is shown in the performance view.

mikakaraila, 2021-06-09:
I don´t understand your question.
UaExpert performance view, is there export? What do you want to do with it?

With node-red you can read csv with file node.

Explain more detailed what you want to do...

mikakaraila, 2021-06-09:
UaExpert does not provide export, but I tested with 5 nodes performance (just curiosity).

[image]

mikakaraila, 2021-06-09:
I will close this as it is not directly related to node-red-contrib-opcua.

closed 2021-06-09 by mikakaraila

ankita1909, 2021-06-10:
> UaExpert does not provide export, but I tested with 5 nodes performance (just curiosity).
> 
> [image]

The values you got using the 5 nodes, is there a way to get these values in text file or any other file. Instead of copying these values manually, i wanted to know if uaexpert write these values in a temp file and i could access it?