mikakaraila/node-red-contrib-opcua #311: Measuring the handshake time in OPC UA

ankita1909, 2021-06-09:
How can i measure the exact handshake time of OPC UA server and also the time it takes to verify the certificates. ?

mikakaraila, 2021-06-09:
Use samples provide by node-opcua.

Then you can use console to track timings.

console.time("Time this");

for (var i = 0; i < 10000; i++) {
  // Your stuff here
}

console.timeEnd("Time this");

But much more easier is to use environment variables to see actual low level messages:

SET NODEOPCUADEBUG="SERVER{TRACE-REQUEST-RESPONSE-STATS-PERF}"

mikakaraila, 2021-06-09:
And there will be difference first time connect and second time connect.
Certificate manager will store certificate first (trusted/rejected) thus second time will be faster.

mikakaraila, 2021-06-09:
You must implement this other way. 

Node-red nor node-red-contrib-opcua cannot be used to make low level performance mesurements.

closed 2021-06-09 by mikakaraila