mikakaraila/node-red-contrib-opcua #312: File: users.json not found!

anxuae, 2021-06-10:
Hello,

When I'm starting node-red (on Windows), I have always the following error, how can I avoid it?
Note that I have not defined a `users.json` in the OPC-UA server node, and it is done voluntarily because I don't use users.

Correct me if I'm wrong, but the normal behavior should be silently use an empty users map if no file defined?

```
File: users.json not found! You can inject users to server or add file to current folder: ~\.node-red\node_modules\node-red-contrib-opcua\opcua
10 Jun 11:46:58 - [error] [OpcUa-Server:LocalServer] File: users.json not found! You can inject users to server or add file to current folder: ~\.node-red\node_modules\node-red-contrib-opcua\opcua
```

mikakaraila, 2021-06-10:
Ack easy to change as optional and not to give error / warning.

mikakaraila, 2021-06-10:
OK if node doesn´t contain users.json filename then it is not loaded / tried to load and no error messages.
[image]

But if it is given then it is loaded:
[image]

Wrong filename, debug tab:
[image]

Console:
[image]

mikakaraila, 2021-06-10:
I made wew version v0.2.224

closed 2021-06-10 by mikakaraila