mikakaraila/node-red-contrib-opcua #313: register_server_manager: contacting discovery server backoff failed

anxuae, 2021-06-10:
Hello,

I have the following information which is repeatedly written in the console where I have started node-red.
Could you tell me what does it means, and what I have done wrong?

```
09:47:04.991Z :register_server_manager       :316   contacting discovery server backoff  opc.tcp://DESKTOP-GBIB5PC:4840  attempt # 1  retrying in  4  seconds
09:47:11.005Z :register_server_manager       :316   contacting discovery server backoff  opc.tcp://DESKTOP-GBIB5PC:4840  attempt # 2  retrying in  8  seconds
09:47:21.021Z :register_server_manager       :316   contacting discovery server backoff  opc.tcp://DESKTOP-GBIB5PC:4840  attempt # 3  retrying in  16  seconds
09:47:39.058Z :register_server_manager       :316   contacting discovery server backoff  opc.tcp://DESKTOP-GBIB5PC:4840  attempt # 4  retrying in  32  seconds
```

mikakaraila, 2021-06-10:
By default server tries to register itself to LDS (Local Discovery Server). I can add option to disable it.

anxuae, 2021-06-10:
Thanks, or you can also put the message with debug level?
or both 😄

mikakaraila, 2021-06-10:
What do you mean? Message comes actually from the node-opcua and it sends those messages. I cannot influence those...

mikakaraila, 2021-06-10:
It seems that I already implemented this some versions ago.

Just turn off Register to local discovery server option:
[image]

closed 2021-06-10 by mikakaraila