mikakaraila/node-red-contrib-opcua #314: OPC Write Takes Forever Sometimes

paulsalibe, 2021-06-10:
I have a bunch of things writing to a WRITE Client node in Node-Red. They are, for the most part, not writing all of the time, but only when a button on the dashboard is pushed, etc. About 40% of the time, when I hit the unlock button for example, it takes up to 20 seconds to unlock the door. Other times it is immediate (as it should be). What could be the cause of this?

mikakaraila, 2021-06-14:
Multiple reason:
1) Other writes on queue to server
2) Is server processing those async read & write with backend server
3) Communication to door WiFi & protocol

There are a lot possible bottlenecks that can cause delay. You can try to register nodeId to server if it can run action faster with registered NodeId.

mikakaraila, 2021-06-14:
Can add some debug output to flow / chain of events like:
1) client write (timestamp) -> 2) Server run action to open door (timestamp) -> 3) Zigbee/mqtt message to door (timestamp)

paulsalibe, 2021-06-18:
Can you explain how the process of registering to nodeId works? Also I received this message when checking the status of node-red in here. 
[image]

mikakaraila, 2021-07-17:
It is actually depending how server is implemented, but it will store nodeId to faster access to original value source.
Here is OPC UA specification:
https://reference.opcfoundation.org/v104/Core/docs/Part4/5.8.5/

mikakaraila, 2021-07-17:
As single writes will be handled as own messages they will go to queue and thus take more time depending how server will handle them. But I expect problem is somewhere else than OPC UA level.

closed 2021-07-17 by mikakaraila