mikakaraila/node-red-contrib-opcua #315: Read node attributes: DisplayName & Description

anxuae, 2021-06-14:
Hi, 

Is it possible to read the some standard attributes of a node like the DisplayName or Description?

mikakaraila, 2021-06-14:
I expect client node INFO should return those:

[image]

anxuae, 2021-06-15:
Hi, yes this is something like that I looking for.

anxuae, 2021-06-17:
It is currently what the client node returns:

[image]

mikakaraila, 2021-06-17:
You must use action INFO in the client node then it will read all attributes from the input message nodeId.

anxuae, 2021-06-17:
thanks, I have not understood that

closed 2021-06-17 by anxuae