mikakaraila/node-red-contrib-opcua #316: Browse nodes opens connection with server and never closes unlike client node

premalatha-b, 2021-06-15:
Hi,
I see that Browse nodes opens connection with server and never closes unlike client node.
Session gets open and closed in browse node. But connection with server remains as is, without closing that connection.
Which creates in increase in number of TCP connection open gradually, with number of calls done to the browse node.

Could you please suggest on it? Will the code base be updated or we can have any alternative?
I didn't find good place to raise this concern. so mentioned here. Sorry for it.

mikakaraila, 2021-06-15:
I will check it and add close session. Thanks for reporting.

mikakaraila, 2021-06-15:
I am bit confused:

Browse node opens & closes session, status is Done after everything is done:
[image]
I added here one line: client.disconnect() perhaps this was your case?

Client node reuses session (keeps it open), node status is browse done:
[image]

mikakaraila, 2021-06-15:
I pushed code to GIT you can test with it. I didn´t yet publish new version.

premalatha-b, 2021-06-16:
Thank you sooo much..

premalatha-b, 2021-06-16:
> I am bit confused:
> 
> Browse node opens & closes session, status is Done after everything is done:
> [image]
> I added here one line: client.disconnect() perhaps this was your case?
> 
> Client node reuses session (keeps it open), node status is browse done:
> [image]

yes, that was my concern

mikakaraila, 2021-06-16:
OK, I will publish new version later.

I look a bit more client node if it could use similar approach. Subscription needs session, but other commands could use on demand client & session. It will save resources on server side.

mikakaraila, 2021-06-18:
v0.2.225 should fix this, reopen if you find something.

closed 2021-06-18 by mikakaraila