mikakaraila/node-red-contrib-opcua #317: Increased number of connection to server after network fail

gtmoyle, 2021-06-16:
Hi 
I am having issues with the following system and would be grateful if you could give me some advice. 

OPC UA Server	Siemens S7 1500 CPU1511
OPC Client 1		Siemens IOT2050 Node Red V1.0.2 OPC UA version 2.224 (192.168.1.115)
OPC Client 		Raspberry PI 4 Node Red V1.0.6 OPC UA version 2.224 (192.168.1.90)

The clients are reading from and writing to the CPU1511.

Under normal conditions the system works well and the connection diagnostic screen for the CPU1511 shows that each client has 2 connections.

Currently the clients are reading 6 variables and writing 1 variable every second.

[image]

The issue I have is when I disturb the network.
•	Power Cycle the PLC
•	Remove a network Cable for more than 30 seconds
When the system restores the number of connections to the OPC UA server from the clients increases and the clients may fail to reconnect. The image below is after a Power Cycle.

[image]

The number of connections to the OPC UA server on the CPU at this point changes often but the only way to get back to 2 connections for each client is to restart node red. (I have been restarting each client.) The CPU1511 has a maximum number of connections 

I have a similar issue when downloading a software change to the PLC. The download process stops the opc server for a second and when the server is restored another connection will be added for each client. I have managed to work round this issue by switching off the poling of variables in node red, while I make the change.

Note. I have the same issues when running node red on a Windows system.

Hope you can help.

mikakaraila, 2021-06-16:
Each OPC UA client node will use 1 session at the moment, you have 2 it uses one for the subscription or reading those 6 variables and another session is for writing. After network break node-opcua will restore session and subscriptions and can use extra session for a while. But server should cleanup not used sessions after timeout period. I have got some similar cases as variable amount is close to operation limit after network break. Siemens advice was to increase limit.

gtmoyle, 2021-06-17:
Thank you for your prompt response to my query.

I have now raised this issues with Siemens.

connection to server after network fail (#317)

mikakaraila, 2021-06-17:
I already had support case with Siemens and solution was: update latest firmware + operation limits to max. Worked in my case.

mikakaraila, 2021-06-17:
Hmm as I rethink it could be the current client node creates new client it doesn´t use session restore.
I have to check my code and starts to use session.on("restored_session", function() {...} would be ideal way to handle after network/communication break setup queue and other stuff in order without doing reset to client.
Current implementation is before session restore was implemented into the node-opcua.

mikakaraila, 2021-06-17:
I cleaned code a bit more, I will test with S7-1500. I will publish new version after tests.

mikakaraila, 2021-06-18:
Improved node status:
[image]

mikakaraila, 2021-06-18:
v0.2.225 should fix this, reopen if you find something.

closed 2021-06-18 by mikakaraila

gtmoyle, 2021-06-18:
Thank you for fixing this.

I will install the update later today,

Sent from MailDroid

Closed #317.

https://github.com/mikakaraila/node-red-contrib-opcua/issues/317#event-4908625047