mikakaraila/node-red-contrib-opcua #318: Automatically accept unknown certificates option is enabled regardless of setting

BraydenVC, 2021-06-22:
Hello, this is my first time writing an issue on GitHub, so please forgive any mistakes, or if this issue has already been raised.

I am using this package for a project that I am working on, and noticed that the "Auto Accept Unknown Certificates" option on the OPC UA Server Node appears to do nothing: Regardless of whether it is enabled or disabled, unknown client certificates will be automatically accepted. I removed all client certificates from PKI/trusted/ and PKI/rejected/ and only Sign&Encrypt with Basic256-Sha256 is enabled on the server, however, even when the setting is disabled, any new client certificates are automatically accepted and placed in PKI/trusted/ .

I looked at 104-opcuaserver.js and noticed that other than the line
`this.autoAcceptUnknownCertificate = n.autoAcceptUnknownCertificate;`
this variable is never mentioned again.
Additionally, in utils.js, the Certificate Managers appear to be instantiated with
`automaticallyAcceptUnknownCertificate: true`
no matter what.

For my project, I modified the 104-opcuaserver.js file lines 195 and 196 to:
`const serverCertificateManager = createCertificateManager(node.autoAcceptUnknownCertificate);`
`const userCertificateManager = createUserCertificateManager(node.autoAcceptUnknownCertificate);`

And I modified the function definitions in utils.js:
`function createCertificateManager(autoAccept) {` <=
`    if (_g_certificateManager) {`
`        return _g_certificateManager;`
`    }`
`    return _g_certificateManager = new opcua.OPCUACertificateManager({`
`        name: "PKI",`
`        rootFolder: path.join(config, "PKI"),`
`        automaticallyAcceptUnknownCertificate: autoAccept` <=
`    });`
`}`
`let _g_userCertificateManager = null;`
`function createUserCertificateManager(autoAccept) {` <=
`    if (_g_userCertificateManager) `
`     return _g_userCertificateManager;`
`    return _g_userCertificateManager = new opcua.OPCUACertificateManager({`
`        name: "UserPKI",`
`        rootFolder: path.join(config, "UserPKI"),`
`        automaticallyAcceptUnknownCertificate: autoAccept` <=
`    });`
`}`

This solved the issue for me. Toggling the setting in node-RED and deploying will toggle whether the server trusts unknown certificates.

mikakaraila, 2021-06-23:
Thanks I will take a look. Using global (store value) inside util.js is not a good solution if you run multiple server nodes.

BraydenVC, 2021-06-23:
Thanks for the fast response.

mikakaraila, 2021-06-24:
Actually your solution is OK, I fixed code in the same way :-)

mikakaraila, 2021-06-24:
Published v0.2.226 that contains fix

AcacioRafael, 2024-04-02:
I'm still having this issue on 0.2.323 version. Even with "Auto Accept Unknown Certificates" the OPC UA server will always accept unknown client certificates and store them in its trusted certificates folder.

mikakaraila, 2025-10-02:
I will try to check next time...