mikakaraila/node-red-contrib-opcua #319: How to use read multiple / write multiple

mikakaraila, 2021-06-28:
I added example, look this:
Flow:
[image]
Items in normal way:
[image]
Client action:
[image]

And import inject with Topic: readmultiple
[image]

Write multiple same way.

mikakaraila, 2021-06-28:
I refactored writing to work similar way as readmultiple (this is not yet fully tested / pushed to git / published):
[image]

mikakaraila, 2021-06-29:
New version 0.2.227 is now available

OriolFM, 2021-11-15:
Hello, thanks for this useful node.

I was checking the example, and I managed to trigger a multiple read. The output, however, is a series of separate messages for each parameter, as if I read them sequentially, but with no index that can be used with the Join node afterwards.

Is this the normal behaviour?

mikakaraila, 2021-11-15:
Did you check topic of the msg?

OriolFM, 2021-11-15:
When I trigger the read multiple, I get 11 messages that look like this:
[image]

On single reads, I tried setting up other message fields before injecting the read (like msg.variable_name) but it didn't work.

The closest solution to what I need has been to use a string node to replace the substrings "ns=2;s=Machine." and ".Value" from msg.topic.nodeId, and then replace each address (for instance, M01MA01MH01#0_E_B for the appropriate variable name, "machine_started" in this case").

[image]

Then, using Join and knowing the total number of variables, I can get an object with the variable names.
[image]

Still, it looks unnecessary convoluted.

It would be more straightforward to just the Name field of the Item node to name the variable after the reading is triggered, that could be used as key in a key/value pair on the payload (instead of pushing a single value as payload). Something like this:
[image]

OriolFM, 2021-11-15:
I've also seen that with Read Multiple, when I try to read more than once, I get an error.

What is working for me is to clear the nodeId array, add all the nodeId again, and then send the read multiple each time I want to read. Otherwise, if I send the Read Multiple twice, I get the following error:
[image]

mikakaraila, 2021-11-15:
Hmm, perhaps you addVariable multiple times?

You should use inject "readAll" to read them again. Ii will read stored nodeIds again and each message contains topic with nodeId and payload contains value.

[image]

mikakaraila, 2021-11-15:
Something like this?

[image]

Function node will use context as save each variable and then combine them to newmsg.

var variableName = msg.topic.nodeId.toString();
// substring(7); // Part after ns=3;i= or ns=1;s=
variableName = variableName.substring(7);
if (context.get(variableName) === undefined) {
    context.set(variableName, msg.payload.toString())
}
console.log("Variable: " + variableName + " value:" +msg.payload.toString())

// Use # as delimeter
if (context.get("variables") === undefined) {
    context.set("variables", variableName)
    variables=variableName;
}
else {
    variables = context.get("variables") + "#" + variableName; // Append
}
if (context.get("values") === undefined) {
    context.set("values", msg.payload.toString());
    values=msg.payload.toString();
}
else {
    values = context.get("values") + "#" + msg.payload.toString(); // Append
}

names = variables.split("#");
vals  = values.split("#");

console.log("Names:" + JSON.stringify(names));
console.log("Vals :" + JSON.stringify(vals));
var newmsg = {};
newmsg.payload = {};
newmsg.payload.variables = names;
newmsg.payload.values = vals;

if (names.length == 2) {
    return newmsg;
}

OriolFM, 2021-11-16:
> I tested inject "readAll" twice to read them again. No problems. Do you inject anything else to client node?

My Clear node array is injected once upon deploying/restarting the flows. Then the inject nodes inject the variables once, automatically.

I didn't know if you had an internal queue, so I gave them 500 ms between injects to make sure the node can process each message before sending the next one.

My server currently is down due to power issues, but I'll share the flow when it's up again.

mikakaraila, 2021-11-16:
OK, I will check what happens with clear array if it will cause something.

mikakaraila, 2021-11-21:
Works, my flow gives error message:

[image]

OriolFM, 2021-11-25:
I've been trying, I still have the same issue.
After a clean restart:
[image]
As you see, the first read works, but on the second it throws an error.

This is the normal output log:
`Connection options:[{"securityPolicy":"1","securityMode":3,"clientCertificateManager":"2","requestedSessionTimeout":300000,"clientName":"3","endpointMustExist":false,"defaultSecureTokenLifetime":200000,"connectionStrategy":"4","keepSessionAlive":true},"http://opcfoundation.org/UA/SecurityPolicy#Basic256",{"untrustUnknownCertificate":true,"state":4,"folderPoolingInterval":5000,"_watchers":"5","_readCertificatesCalled":true,"_filenameToHash":"6","_thumbs":"7","_pending_crl_to_process":0,"queue":"8","location":"9","keySize":2048,"referenceCounter":0,"automaticallyAcceptUnknownCertificate":true},"SCHMID Preclean 1",{"maxRetry":10512000,"initialDelay":5000,"maxDelay":30000},[],{"/root/.config/node-red-opcua-nodejs/PKI/trusted/certs/NodeOPCUA-Client@noderedu1.unimicron.int[a91241125f5aaec7002578f332ea6d92fc7b681b].pem":"10","/root/.config/node-red-opcua-nodejs/PKI/trusted/certs/SchmidOpcUaServer[d04bbb7ded9be76324a2e6b8d0a8502fe42edd1a].pem":"11"},{"rejected":"12","trusted":"13","issuers":"14","crl":"15","issuersCrl":"16"},[],"/root/.config/node-red-opcua-nodejs/PKI","a91241125f5aaec7002578f332ea6d92fc7b681b","d04bbb7ded9be76324a2e6b8d0a8502fe42edd1a",{},{"a91241125f5aaec7002578f332ea6d92fc7b681b":"17","d04bbb7ded9be76324a2e6b8d0a8502fe42edd1a":"18"},{"certs":"19"},{},{},{"certificate":"20","filename":"21"},{"certificate":"22","filename":"23"},{},{"type":"24","data":"25"},"/root/.config/node-red-opcua-nodejs/PKI/trusted/certs/NodeOPCUA-Client@noderedu1.unimicron.int[a91241125f5aaec7002578f332ea6d92fc7b681b].pem",{"type":"24","data":"26"},"/root/.config/node-red-opcua-nodejs/PKI/trusted/certs/SchmidOpcUaServer[d04bbb7ded9be76324a2e6b8d0a8502fe42edd1a].pem","Buffer",[48,130,4,126,48,130,3,102,160,3,2,1,2,2,20,47,199,13,103,105,130,76,188,130,8,10,18,162,21,237,118,208,185,249,9,48,13,6,9,42,134,72,134,247,13,1,1,11,5,0,48,101,49,11,48,9,6,3,85,4,6,19,2,70,82,49,16,48,14,6,3,85,4,7,12,7,79,114,108,101,97,110,115,49,17,48,15,6,3,85,4,10,12,8,83,116,101,114,102,105,118,101,49,49,48,47,6,3,85,4,3,12,40,78,111,100,101,79,80,67,85,65,45,67,108,105,101,110,116,64,110,111,100,101,114,101,100,117,49,46,117,110,105,109,105,99,114,111,110,46,105,110,116,48,30,23,13,50,49,49,49,49,53,48,57,52,52,49,50,90,23,13,51,49,49,49,49,51,48,57,52,52,49,50,90,48,101,49,11,48,9,6,3,85,4,6,19,2,70,82,49,16,48,14,6,3,85,4,7,12,7,79,114,108,101,97,110,115,49,17,48,15,6,3,85,4,10,12,8,83,116,101,114,102,105,118,101,49,49,48,47,6,3,85,4,3,12,40,78,111,100,101,79,80,67,85,65,45,67,108,105,101,110,116,64,110,111,100,101,114,101,100,117,49,46,117,110,105,109,105,99,114,111,110,46,105,110,116,48,130,1,34,48,13,6,9,42,134,72,134,247,13,1,1,1,5,0,3,130,1,15,0,48,130,1,10,2,130,1,1,0,219,65,240,41,108,144,50,102,155,210,186,21,11,169,147,173,210,5,98,120,203,29,20,190,178,194,91,228,146,9,41,49,24,167,70,2,72,53,69,251,101,247,148,56,184,207,10,85,250,134,67,72,152,191,115,248,127,57,207,30,23,33,232,203,253,209,185,70,214,226,190,94,136,69,107,101,111,8,116,206,57,49,151,106,89,69,45,81,209,100,31,103,203,208,44,21,226,53,38,193,101,55,105,131,15,125,164,247,51,99,142,185,23,167,86,183,5,66,180,217,198,139,121,228,57,161,153,48,59,119,82,254,79,189,43,205,55,125,55,88,106,86,25,141,231,85,71,114,131,242,56,254,25,1,209,39,92,60,133,54,111,243,249,32,228,25,26,217,115,121,17,78,119,139,110,82,246,155,129,250,181,61,235,223,247,234,224,16,67,169,121,23,13,59,59,119,49,209,79,183,115,94,5,206,108,112,236,84,24,228,35,54,26,153,177,159,239,164,89,34,241,27,7,182,177,178,161,109,212,149,13,105,43,102,233,3,111,9,161,190,17,139,247,239,139,147,222,185,97,245,117,34,83,211,32,145,2,3,1,0,1,163,130,1,36,48,130,1,32,48,29,6,3,85,29,14,4,22,4,20,204,187,178,56,138,166,81,54,229,123,18,85,165,126,18,41,59,194,156,243,48,31,6,3,85,29,35,4,24,48,22,128,20,204,187,178,56,138,166,81,54,229,123,18,85,165,126,18,41,59,194,156,243,48,12,6,3,85,29,19,1,1,255,4,2,48,0,48,11,6,3,85,29,15,4,4,3,2,2,244,48,29,6,3,85,29,37,4,22,48,20,6,8,43,6,1,5,5,7,3,2,6,8,43,6,1,5,5,7,3,1,48,82,6,9,96,134,72,1,134,248,66,1,13,4,69,22,67,83,101,108,102,45,115,105,103,110,101,100,32,99,101,114,116,105,102,105,99,97,116,101,32,103,101,110,101,114,97,116,101,100,32,98,121,32,78,111,100,101,45,79,80,67,85,65,32,67,101,114,116,105,102,105,99,97,116,101,32,117,116,105,108,105,116,121,48,80,6,3,85,29,17,4,73,48,71,134,44,117,114,110,58,110,111,100,101,114,101,100,117,49,46,117,110,105,109,105,99,114,111,110,46,105,110,116,58,78,111,100,101,79,80,67,85,65,45,67,108,105,101,110,116,130,23,110,111,100,101,114,101,100,117,49,46,117,110,105,109,105,99,114,111,110,46,105,110,116,48,13,6,9,42,134,72,134,247,13,1,1,11,5,0,3,130,1,1,0,54,35,125,229,236,121,148,128,197,188,30,175,132,55,215,108,35,219,167,51,142,254,31,202,136,149,28,135,22,251,143,150,62,239,181,168,31,81,196,81,253,145,208,169,233,160,27,204,29,188,119,172,73,215,65,199,181,141,144,152,155,73,85,109,27,87,76,56,173,36,244,18,185,60,81,255,115,182,237,217,55,110,33,63,236,232,24,123,37,153,152,216,153,69,140,223,70,207,31,148,240,244,51,120,191,168,171,137,205,55,70,52,19,160,112,84,144,156,185,244,155,171,100,4,112,226,148,93,105,65,40,67,226,231,165,153,246,26,56,222,6,207,139,116,174,121,46,211,65,58,156,26,64,137,35,143,23,221,17,239,17,10,56,204,146,96,40,97,99,255,157,104,98,127,141,135,15,198,27,145,217,190,104,164,34,53,166,139,121,229,120 [... cut ...]
    at doSQL (/root/.node-red/node_modules/node-red-contrib-mssql-plus/src/mssql.js:779:25)
    at /root/.node-red/node_modules/node-red-contrib-mssql-plus/src/mssql.js:765:17
    at runMicrotasks (<anonymous>)
    at runNextTicks (internal/process/task_queues.js:60:5)
    at processImmediate (internal/timers.js:437:9) {
  code: 'ECONNCLOSED'
}
(node:6248) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 acquire listeners added to [Pool]. Use emitter.setMaxListeners() to increase limit
`

OriolFM, 2021-11-25:
If I clean the reading queue everytime, add all the nodes, and trigger the multiple read, it works. Otherwise, it doesn't.

mikakaraila, 2021-11-25:
But error seems to be in the mssql interface?

ConnectionError: Connection is closed. at Request._query (/root/.node-red/node_modules/mssql/lib/base/request.js:495:37) at Request._query (/root/.node-red/node_modules/mssql/lib/tedious/request.js:367:11) at /root/.node-red/node_modules/mssql/lib/base/request.js:459:12 at new Promise (<anonymous>) at Request.query (/root/.node-red/node_modules/mssql/lib/base/request.js:458:12) at connection.node.execSql (/root/.node-red/node_modules/node-red-contrib-mssql-plus/src/mssql.js:423:40) at doSQL (/root/.node-red/node_modules/node-red-contrib-mssql-plus/src/mssql.js:779:25) at /root/.node-red/node_modules/node-red-contrib-mssql-plus/src/mssql.js:765:17 at runMicrotasks (<anonymous>) at runNextTicks (internal/process/task_queues.js:60:5) at processImmediate (internal/timers.js:437:9) { code: 'ECONNCLOSED' }

OriolFM, 2021-11-30:
My OPC-UA flow is completely unrelated to the mssql interface, it is not connected to it and it does not depend on it.

If there is a problem with the MSSQL, it is from another group of machines, that supplier made a mess with their DB. In any case, I still can read it.

The problem I have now is that I do not have an efficient way of reading multiple variables from the OPC-UA interface for these other production lines. In order to do so, I still have to clear and initialize the variables every single time.

OriolFM, 2021-11-30:
I mean, after my second inject with "readmultiple", my catch error node focused on the OPCUA client node generates an output where the error message is:
`Error: String cannot be coerced to a nodeId : machine_started
    at coerceNodeId (/root/.node-red/node_modules/node-opcua-nodeid/dist/nodeid.js:250:19)
    at resolveNodeId (/root/.node-red/node_modules/node-opcua-nodeid/dist/nodeid.js:343:18)
    at ClientSessionImpl.read (/root/.node-red/node_modules/node-opcua-client/dist/private/client_session_impl.js:542:65)
    at ClientSessionImpl.read (/root/.node-red/node_modules/thenify/index.js:65:46)
    at ClientSessionImpl.read (/root/.node-red/node_modules/node-opcua-client/dist/private/client_session_impl.js:515:25)
    at ClientSessionImpl.read (/root/.node-red/node_modules/thenify/index.js:65:46)
    at readmultiple_action_input (/root/.node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:730:22)
    at OpcUaClientNode.processInputMsg [as _inputCallback] (/root/.node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:538:11)
    at /usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/...`

mikakaraila, 2021-11-30:
Hmm cannot check code line as you have now older version running (102-opcuaclient.js: 730).

String machine_started is not valid nodeId as it doesn´t have ns=1;s=

Can you install latest version to check code and rerun to get correct code line into the error message?

OriolFM, 2021-11-30:
machine_started is the name of the node where the error occurs:

[image]

It is also the first variable in the list. I already updated to the last version and same thing happens. This is the output of the catch error node in node-red:
`{"_msgid":"59beee0d36730a37","payload":"","topic":"readmultiple","error":{"message":"Error: String cannot be coerced to a nodeId : machine_started","source":{"id":"95c94d51.0c6be","type":"OpcUa-Client","name":"SCHMID Preclean 1","count":1},"stack":"Error: String cannot be coerced to a nodeId : machine_started\n    at coerceNodeId (/root/.node-red/node_modules/node-opcua-nodeid/dist/nodeid.js:257:19)\n    at resolveNodeId (/root/.node-red/node_modules/node-opcua-nodeid/dist/nodeid.js:350:18)\n    at ClientSessionImpl.read (/root/.node-red/node_modules/node-opcua-client/dist/private/client_session_impl.js:542:65)\n    at ClientSessionImpl.read (/root/.node-red/node_modules/thenify/index.js:65:46)\n    at ClientSessionImpl.read (/root/.node-red/node_modules/node-opcua-client/dist/private/client_session_impl.js:515:25)\n    at ClientSessionImpl.read (/root/.node-red/node_modules/thenify/index.js:65:46)\n    at readmultiple_action_input (/root/.node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:818:22)\n    at OpcUaClientNode.processInputMsg [as _inputCallback] (/root/.node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:541:11)\n    at /usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/..."}}`

mikakaraila, 2021-11-30:
Looks for me that flow mixes msg.topic and msg.name can you check your flow?

OriolFM, 2021-11-30:
Again, this is my flow:
[image]

This is the full message I inject to add the nodeID:
[image]

This is what I inject to read the multiple values:
[image]

After I get the messages, I replace each nodeID by the variable name using the String node, then I join all the messages, and I get the following output:
[image]

If then I inject the same "readmultiple" message as I did, I get the error again. See the debug output:
[image]

mikakaraila, 2021-11-30:
In this msg topic is correct:

[image]

Here is the wrong one (before msg.topic readmultiple):
[image]

OriolFM, 2021-11-30:
> Here is the wrong one (before msg.topic readmultiple): [image]

That is the processed output from the first readmultiple command. In the detailed explanation you can see the output timestamp.

mikakaraila, 2021-11-30:
Can you add to flow debug all that is injected into the client node?
Just to check that msg topic is always nodeId.

OriolFM, 2021-11-30:
Okay, now I have no idea what was wrong.
I simplified the flow, just getting the raw output of the OPC-UA node, and now it's working.
[image]

Since I am injecting exactly the same as before, I have no idea what was wrong in it.
I also tested the "ALL" option as you suggested, and it gives me the array of values and the array of items.

Now I have to figure out how to replace the strings by human-readable variable names. Maybe in a future update, you could add a variable name to the OPC-Item node, that allows the user to specify a string name that can be used later to create a key/value pair?

I'm going to write a function node to get an output like the one I refer to, then another one to combine key/pairs into an object, for a tidier handling. If you're interested, I can share it. I know it's nothing to write home, but the work will be done anyways.

mikakaraila, 2021-11-30:
Ack good that you found it. I have Digital Twin under work. I will do just needed fixes to node-red-contrib-opcua.

OriolFM, 2021-12-07:
Hi, have you considered including the possibility to add several items from one node, a bit like the S7 node does?

You configure each endpoint with the variables you want to read from it, it looks like this:
[image]

Alternatively, there is the option to import/export the endpoints as a csv file, like this one:
`DB13,X20.4;downstream_ready
DB13,X22.0;conv1_in
DB13,X22.1;conv1_slow
DB13,X22.2;conv1_stop
DB13,X22.3;conv1_out
DB13,X22.4;conv1_occupied
DB13,X58.0;conv2_in
DB13,X58.1;conv2_slow
DB13,X58.2;conv2_stop
DB13,X58.3;conv2_out
DB13,X58.4;conv2_occupied
DB13,X94.0;bridge_down_CP
DB13,X94.1;bridge_down_MT
DB13,X94.2;conv3_in
DB13,X94.3;conv3_slow
DB13,X94.4;conv3_stop
DB13,X94.5;conv3_out
DB13,X94.6;conv3_occupied
DB13,X96.0;bridge_REQ_CP
DB13,X96.1;bridge_REQ_MT
DB13,X114.0;bridge_cyl1_locked
DB13,X114.1;bridge_cyl1_unlocked
DB13,X114.2;bridge_cyl2_locked
DB13,X114.3;bridge_cyl2_unlocked
DB207,X4.3;homing
DB207,X4.4;cyclestop
DB207,INT10;opmode
DB231,X0.1;ready
DB2303,X36.3;bridge_open
`

I was thinking that it could be either this, or a JSON string with each variable (whatever format you prefer, but that it was easier to edit). Allowing to assign variable names to the OPC-UA nodes would save us from having to add them like this:
[image]

And then filter the nodes like this:
[image]

At the moment I have one of the machines that requires more than 1000 lines of code only for the if (){}else if ()... to filter all the variables.

Again, if you have an idea on how to do it and you think you could use some help, I don't consider myself a specially good programmer, but collaborating in something like this would also save me some work down the road, so whatever I can do, I will.

mikakaraila, 2021-12-08:
OK, I will take a look to that node if I could add similar node to handle read/write multiple.

mikakaraila, 2022-03-21:
@OriolFM can you explain / give some coordinates for the "Variable list"?

I could implement similar with node crawler. Then it would be easier to select nodeId from the list to read/write.

OriolFM, 2022-03-21:
> @OriolFM can you explain / give some coordinates for the "Variable list"?

The S7 node endpoints have a list where you can already specify which variables you want to read. It looks like this:
[image]

In the past, you had to define them yourself one by one, but in more recent versions they added the Import/export buttons, allowing the endpoint node to read or write CSV files where the memory addresses, variable type, and offset are stored, as well as a variable name, so the PLC node will already produce an output with a human-readable variable name.

This is an example of a CSV file I'm currently using:
```
DB13,X135.0;conv_input
DB13,X135.1;conv_slow
DB13,X135.2;conv_stop
DB13,X135.5;conv_output
DB201,X15.0;PortA_AGV_mode
DB201,X15.1;PortA_BC
DB201,X19.0;PortB_AGV_mode
DB201,X19.1;PortB_BC
DB201,X39.1;DMC_EN
DB207,INT2;opmode
DB208,X0.3;homing
DB208,X0.4;cyclestop
DB2530,X8.1;PortA_EN
DB2530,X8.2;PortB_EN
DB2530,S24.40;DMC
DB2530,INT70;PortA_Count
DB2530,INT74;PortB_Count
DB2550,X0.1;PortA_REQ
DB2550,X0.4;PortA_dock
DB2550,X0.5;PortA_onTheWay
DB2550,X0.6;PortA_sensorL
DB2550,X0.7;PortA_sensorR
DB2550,X1.0;PortA_transfer_busy
DB2550,X1.1;PortA_transfer_done
DB2550,X1.4;PortA_receive_busy
DB2550,X1.5;PortA_receive_done
DB2550,X20.1;PortB_REQ
DB2550,X20.4;PortB_dock
DB2550,X20.5;PortB_onTheWay
DB2550,X20.6;PortB_sensorL
DB2550,X20.7;PortB_sensorR
DB2550,X21.0;PortB_transfer_busy
DB2550,X21.1;PortB_transfer_done
DB2550,X21.4;PortB_receive_busy
DB2550,X21.5;PortB_receive_done
DB9000,S320.4;PortA_ID
DB9000,S418.4;PortB_ID
```

This saves a lot of work, being able to define the variables you read from each endpoint and the meaningful variable name you want to use (as opposed to the address + variable type + offset).

I estimate that about 75% of the time I spent working with the OPC-UA nodes was getting the output from the node, loop through the items array and compare to a whole list of nodes and variable names, and assign the variable name if the item name string contained a specific substring, then getting that output arrays with variable names and convert them to a JSON object sorted by modules (the machine is split in different modules, so I separate the variables for each one of them).

The only issue I have with the S7 node is that the variables you read from an endpoint are fixed. You can choose to read all of them at once, or only produce the values that changed since the last update (similar to a subscription). You can setup a regular interval for reading the variables, or you can use an S7 Control node to trigger an asyncronous read if you want, but each time you'll get either all variables or just the ones that have changed from last time, depending on which mode you selected. You can't just choose to read the value from a specific variable and only that one if you configured 40 entries.

Your OPC-UA node is more versatile in which you can dynamically alter the variables you want to read each time, but sometimes it would be useful to just configure everything in the endpoint node and just trigger the reads whenever you want (or put it in subscription mode and let it generate the output when a value from the list changes). The option of sending a manual request at the node input could still be there if the variable list was empty, so it would keep backwards compatibility, for instance.

anxuae, 2022-10-20:
I @mikakaraila looking in the exemples, I can not find a way to pass an array of nodeId to be read. Is it possible?
Also I would like that thsi array replace the existing one (clear existing item)

mikakaraila, 2022-10-21:
I found this example which subscribes multiple by sending nodeId array.
[MULTISUB-TEST.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/9836024/MULTISUB-TEST.txt)

anxuae, 2022-10-21:
I'm on the latest version on the library, but it does not work, I've got the following error:

it seems that `msg.topic="multiple"` is not supported in READ MULTIPLE (your example is for SUBSCRIBE MULTIPLE)
`
```
Error: String cannot be coerced to a nodeId : multiple
    at coerceNodeId (/data/node_modules/node-opcua-nodeid/dist/nodeid.js:264:19)
    at resolveNodeId (/data/node_modules/node-opcua-nodeid/dist/nodeid.js:358:18)
    at ClientSessionImpl.read (/data/node_modules/node-opcua-client/dist/private/client_session_impl.js:571:65)
    at ClientSessionImpl.read (/data/node_modules/thenify/index.js:65:46)
    at ClientSessionImpl.read (/data/node_modules/node-opcua-client/dist/private/client_session_impl.js:544:25)
    at ClientSessionImpl.read (/data/node_modules/thenify/index.js:65:46)
    at readmultiple_action_input (/data/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:1124:22)
    at OpcUaClientNode.processInputMsg [as _inputCallback] (/data/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:601:11)
    at hooks.trigger (/usr/src/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:203:26)
    at Object.trigger (/usr/src/node-red/node_modules/@node-red/runt...
```

mikakaraila, 2022-10-21:
I KNOW, as I wrote "I found this example which **subscribes** multiple by sending nodeId array."

I just don´t have time to check read multiple example / interface...

OriolFM, 2022-11-07:
> I KNOW, as I wrote "I found this example which **subscribes** multiple by sending nodeId array."
> 
> I just don´t have time to check read multiple example / interface... Subscribe is just better than read.

For subscription, is there any way to receive all the variables that have changed in a single array, instead of receiving n messages? I read am reading several OPC-UA servers with over 200 nodes each and the message overhead is taxing on my node-red server.

Thanks!

mikakaraila, 2022-11-07:
As far as I have experience only read multiple will return all values in one array. 

As subscription is triggered only on value changes those are individual item that will get notification. This is normal OPC UA functionality.

Only as subscription is created it is possible to add all items in array.

OriolFM, 2022-11-08:
Then, what about adding the nodeIds in an array for Read Multiple, like in the Subscription example? Is it possible?

I've been trying to add all the nodeIDs in a single array for my readmultiple application, but it is not working.

At the moment, I have a lot of opc-ua item nodes (between 180-300) to read the data from several chemical production lines. Having so many forked wires increases the memory consumption of node-red and creates a lot of overhead.

The main problem with this line is that almost all of the OPC-UA parameters are analog readings from sensors that change almost every time, so I get about 80-90% of all my parameters every time I read in subscription mode. That generates a lot of messages, and it is actually less costly (regarding memory consumption and speed of processing) to get a single array with read multiple in a single message than getting the subscribed parameters one by one in separate messages.

Also, processing the parameters is easier if you have all of them in a single array and convert it to a JSON object with named properties, that you can use with a simple change node to overwrite the context object that holds the information and you display on the dashboard.

With subscription, the code is more complex, it is necessary to update each single property of the context object separately, requiring more logic, processing power and steps.

OriolFM, 2022-11-10:
Thank you very much! I was wondering about that :)

I'll give it a try when it's ready and let you know how it goes.

mikakaraila, 2022-11-10:
And works :-) I will publish new v0.2.292 in few minutes.

[image]

[mread_prosys.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/9983104/mread_prosys.txt)

OriolFM, 2022-11-11:
> And works :-) I will publish new v0.2.292 in few minutes.
> 
> [image]
> 
> [mread_prosys.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/9983104/mread_prosys.txt)

I tested it, followed these steps:

1.  After deployment, I send the "cleatitems" msg to make sure there's nothing in the node
2.  Inject the NodeID array with readmultiple as topic, as per your example.
                _In a debug node at the output, I see the node is returning one message for each nodeID when it's processing the array._
3. After that, whenever I send a message with "readmultiple" as topic, it will return the whole array as it should.

So actually, I'd say it works fine, except for returning all these individual messages when the array is processed. Either send nothing when the array is processed, or send the normal array once all the nodeIDs have been added.

mikakaraila, 2022-11-11:
What you mean with this comment?

"except for returning all these individual messages when the array is processed"

I don´t have Ko-Fi page, done everything free :-) Just enjoy helping people and solving problems. Kind of hobby for me.

OriolFM, 2022-11-11:
This is the test I set up:
[image]

This is the output I get when I inject the array:
[image]
 As you can see there's an "unknown nodeID" message, then one message per nodeID in the array (in sequential order).
 
 If I inject an empty msg with "readmultiple" as topic, then I get the expected result:
[image]

So actually there are two issues: this "unknown nodeID" error I get in the beginning, and the node producing one message per nodeID when I inject the initial array (it does not return anything if I inject the nodeIDs separately).

Regarding the error message about an unknown nodeID, I am not sure where is the error, since I inject 28 nodeIDs and I get 28 values back in the full array, which means all have been processed.

mikakaraila, 2022-11-11:
Error message comes from the server as it tries to read each nodeId.

I expect you has typo in one of the nodeId´s:
[image]

closed 2023-02-28 by mikakaraila