mikakaraila/node-red-contrib-opcua #320: Monitored items

lukinno1, 2021-06-29:
Hello

How can i, which nodes are monitored?

I used subscribe, it is run but when i used unsubscribe i do not have answer. How can i check it.?

Lukas

mikakaraila, 2021-06-29:
When you make subscription to variable, node will send topic and payload is subscriptionId:
[image]

mikakaraila, 2021-06-29:
Do you have unsubscribe this way:
[image]

mikakaraila, 2021-06-29:
Flow snippet:
[image]

lukinno1, 2021-06-29:
> 
> 
> When you make subscription to variable, node will send topic and payload is subscriptionId:
> [image]

ok i have subcriptionid, but how can i which items are monitored?

lukinno1, 2021-06-29:
> 
> 
> Flow snippet:
> [image]

sent me json import

mikakaraila, 2021-06-30:
You already have it, just look examples folder OPCUA-TEST-NODES.json

closed 2021-07-17 by mikakaraila