mikakaraila/node-red-contrib-opcua #321: Is it possible to get Server Time Stamps along with READ and READ MULTIPLE?

Darwin1986, 2021-06-30:
For some applications it is required to read mutliple variables at the same time and also acquire the correct server timestamp for that. Is there a way to retrieve the timestamps as well on READ or especially READ MULTIPLE?

Currently Read returns just the following elements:
[image]

There is no timestamp included at all. For READ MULTIPLE the answer looks as below. In case READ MULTIPLE is used along with the payload option "ALL" the situation is even worse, since the topic is not included in the final array for each variable as well. So I am not sure how I can be sure that I interpret the data correctly.
[image]

The only operation that returns a server timestamp seems to be a subscribe, which I can't use since I need cyclic reading of all variables at the same time,to get a consistent state of the monitored aggregate. And if the value does not change I do not get new values.

Is there an option to activate the serverTimestamps on READ or READ MULTIPLE?

mikakaraila, 2021-06-30:
I have to check if node-opcua API allows to use timestampsToReturn: TimestampsToReturn.Both,

mikakaraila, 2021-06-30:
I just tested both and read multiple returns null for both, but read will return timestamps:
[image]

mikakaraila, 2021-06-30:
Payload:
[image]

mikakaraila, 2021-06-30:
I added timestamps into the code if server will not return them, this is "workaround" for readmultiple:
[image]

Darwin1986, 2021-06-30:
Where do I need to adapt the code? This wasn't present before right? In my case read does not return any timestamps. I am not sure where to change the option for TimeStampsToReturn. Can you give me a hint?

mikakaraila, 2021-06-30:
I just pushed it to github, published and update to nodered.org
You can just install v0.2.228

mikakaraila, 2021-07-01:
I found that read multiple was using old API call, now it uses correct one and both timestamps are returned correctly.

mikakaraila, 2021-07-01:
v0.2.229 contains fix and it is published

closed 2021-07-01 by mikakaraila