mikakaraila/node-red-contrib-opcua #322: Certificates from Server is auto rejected from version 0.2.226 +

L-uva, 2021-07-01:
Hello,

I have problems when setting a an opcua client with the newer version of this module.
The problem started on .226 version

The certificate that the server is sending is automatically put in the rejected folder.
I can move that cert into the trusted folder and restart node-red and it works perfectly.

I did also downgrade to .225 version to verify that it works on the older version. It worked immediately on the older version.

mikakaraila, 2021-07-01:
Check the setting: "Auto accept unknown certificates". It was earlier always true.

[image]

mikakaraila, 2021-07-01:
This was issue:
https://github.com/mikakaraila/node-red-contrib-opcua/issues/318

L-uva, 2021-07-02:
Yes but thats for the server node, I'm using the client node.

I read about #318 issue, maybe setting that globally wasnt a good idea?

mikakaraila, 2021-07-02:
Yes, you are correct. I will fix this.

mikakaraila, 2021-07-02:
Fixed to v0.2.230

closed 2021-07-02 by mikakaraila