mikakaraila/node-red-contrib-opcua #323: Can't browse all node

Puerto91, 2021-07-02:
Hi,

I am trying to make an application that connects the siemens s7-1500 PLC with nodered via OPC UA

The application should automatically load the "DataBlock" list. 
When i use aplication "Unified automation UaExpert" i've got 3 element when i browse "ns=3;s=DataBlocksGlobal"
[image]

But when i use node OPC UA browser in NodeRed, i've got only one item "Icon"

[image]

You can tell me what I'm doing wrong?

mikakaraila, 2021-07-02:
If those types are ExtensionObjects they are not yet supported. I have been working with ExtensionObjects still some work to do.

mikakaraila, 2021-07-02:
There is another way to browse variables: client node and action browse. Try it!

Puerto91, 2021-07-04:
Thanks you for quick response!
 Client with action browser works, but a little different then i expect.
When i send topic "ns=3;s=DataBlocksGlobal" i've got in output more the 1000 msg. One of this msg have information what i need in reference nodes:)
I've have to use simple filter on outpt and i have all info what i need.

closed 2021-07-04 by Puerto91