mikakaraila/node-red-contrib-opcua #324: No reconnection after the OPC/UA server (the machine) has been switched off and on again

kruegeralex, 2021-07-14:
## In advance:

Super great product, thanks for the work that has been put in so far! 

## Current problem:
The connection established by the node is interrupted after a certain time and does not re-establish itself. 
[image]
The connection can be re-established if the machine is currently switched on and the flow is re-deployed, thus forcing a new connection.

## Assumption:
The connection is probably interrupted by the fact that the machine is switched off and thus also the OPC server and the machine remains switched off longer than the standard timeout (60000ms?).

## Approach:
After the standard timeout time has been exceeded check all (60000ms) if the OPC-Server is reachable again and so the connection can be re-established.

mikakaraila, 2021-07-15:
Node-opcua contains re-connect, any more details?

kruegeralex, 2021-07-15:
We just use the node on machines that are switched on and off frequently. Whenever the machine is switched off, the connection to the node is interrupted (logically). As soon as the machine is switched on again, the node does not try to reconnect and the error message above appears. 
You can force a reconnect by deploying the flow again.

Deployed is Nodered via a Helm Chart on an x86 Edge-PC.

We already know a similar problem from Kepware.

The reconnect is limited to 60000ms isn't it?

mikakaraila, 2021-07-15:
There's max retry, how long machine is turned off?

mikakaraila, 2021-07-15:
Client node will show status this way:
[image]

mikakaraila, 2021-07-15:
These parameters are now in the code:
connectionOption.connectionStrategy = {
      maxRetry: 10, // Limited to max 10 ~5min.
      initialDelay: 5000, // 5s
      maxDelay: 30 * 1000 // 30s
    };

mikakaraila, 2021-07-15:
So if you turn off server longer than 5min time then maxRetry must be much bigger.

mikakaraila, 2021-07-15:
Code seems to retry if connection is once created even more than given maxRetry count:
[image]

Please check first your node-red-contrib-opcua version and update it to the latest one v0.2.230

kruegeralex, 2021-07-16:
The machines sometimes stay with our users for several days or even a week. This actually happened more often with other users, for example at Christmas or during company vacations.

Would it be possible to include an option in the node where the node tries to reconnect at all times without limit? I think there is a relatively big case for this as many/most of our users turn off the machine from time to time e.g. for maintenance and then want everything to reconnect as usual.

mikakaraila, 2021-07-16:
Ok I will test how many days it will survive.

mikakaraila, 2021-07-17:
As testing takes days, I already changed maxRetry to ~10 years time. Feel free to test.
Provide some console output or log what happens in case of re-connect will not work.
New version will be v0.2.231 install it before testing!

kruegeralex, 2021-07-20:
We have it in production testing on a machine now.
I'll keep you posted.

Thank you so far!

ShaheryarAhmad-IIOT, 2021-07-24:
@kruegeralex 

Sorry...But, happy to hear that I have faced the same issue and as you are having same...Then, this thing can be resolved by collectives work :)

Well, for intro I have a production machine in which 80 tags are enabled over OPCUA. My Node-RED works smooth but after machine off, it gets the same error...

So, What's the progress now...Have you found any satisfied result?

mikakaraila, 2021-07-24:
I need console log or something to look...

ShaheryarAhmad-IIOT, 2021-07-24:
@mikakaraila 

I have updated my OPCUA node to 0.2.233...

So, as aboved mentioned you have changed the max retry to 10 years...
Will this 0.2.233 has the maxretry to 10 years?

mikakaraila, 2021-07-24:
Yes and I wonder what error can be seen on console?

ShaheryarAhmad-IIOT, 2021-07-25:
@mikakaraila 

Thank You for responding at your earliest...

I will update the OPCUA node tomorrow (Monday) and will test it for some days... Hopefully, this update will solve the issue. As, I am facing the same issue as mentioned in this thread's subject.

Will update you asap ✌️