mikakaraila/node-red-contrib-opcua #325: client-write does not returns opc ua status

anandanatarajan, 2021-07-16:
hi,
how to check whether the write has been done and good , i tried to write to kepserver via opc ua client write, it writes data to kepserver but it does not return any status, we have one flow which depends on status codes return from write. is it possible or is there any other way to do it.

mikakaraila, 2021-07-16:
Ok I have just forgotten to send topic and status as payload.

anandanatarajan, 2021-07-17:
hi,
thanks for the quick response, will there be any update in that regard , if that is too much too ask, then can you suggest any workaround

mikakaraila, 2021-07-17:
I just added two lines to send msg with payload that contains statuscode. v0.2.231 will be available in minutes.

anandanatarajan, 2021-07-21:
Hi, 
Thanks for the very quick response and update, just one question, sorry if it is stupid question
```
ConstantStatusCode {
  _value: 0,
  _description: 'No Error',
  _name: 'Good'
}
```
this is the output i am getting from the write method, it is not validating as a json, how do i process this output any inputs
thanks once again.

mikakaraila, 2021-07-21:
You have to convert object to JSON and parse it, use function node.
Here is short example:

var status = msg.payload.toJSONFull(); // ConstantStatusCode object with properties value, description and name
console.log("status=" + JSON.stringify(status));

var status2 = JSON.parse(JSON.stringify(status)); // Parse back to normal object
console.log("value= " + status2.value); // Access value property

mikakaraila, 2021-07-21:
Flow and after write node function node:
[image]

mikakaraila, 2021-07-21:
I am just fixing statuscode, the current code does not return actual statuscode after write.

mikakaraila, 2021-07-21:
Seems to work now with v02.233

I added test cases: 1) Try to write read only access node 2) Try to write wrong nodeId

[image]

closed 2021-07-27 by mikakaraila