mikakaraila/node-red-contrib-opcua #329: OPC UA Client node: error by 1 in node id for READ function

Girgitt, 2021-07-19:
*Environment:*
Centos 7 AMD64, 

node-red version:
Welcome to Node-RED
19 Jul 15:14:48 - [info] Node-RED version: v1.1.3
19 Jul 15:14:48 - [info] Node.js  version: v12.20.2
19 Jul 15:14:48 - [info] Linux 3.10.0-1127.19.1.el7.x86_64 x64 LE

node-red-contrib-opcua version: 0.2.231

*steps to reproduce:*
1. create test OPC UA server where values depend on given node's id
2. configure OPC UA Client node, point it to test server and inject messages with ns=2;i=X

*expected behavior:*
1. for injected msg.topic: ns=2;i=1 client returns value for node msg.topic=ns=2;i=1
2. for injected msg.topic: ns=2;i=2 client returns value for node msg.topic=ns=2;i=2

*actual behavior:*
1. for injected msg.topic=ns=2;i=1 client returns Null
2. for injected msg.topic=ns=2;i=2  client returns value of ns=2;i=1 (confirmed with UA Expert client)

mikakaraila, 2021-07-19:
Did you test with multiple variables?
Code uses actually ns=1;s=browseName syntax not index based

Girgitt, 2021-07-20:
Thank you for your work and support and sorry for bothering you with what turned out to be a false issue report. 

More tests **do not confirm the initially reported issue** - most probably I forgot to refresh the UA Expert client's property panel after more tags were registered for polling. 

The "ns=X;s=browseName" syntax does not seem to work with a custom opc server based on opcua python module but "ns=X;i=Y" syntax works fine. I will go through all reported issues and available documentation in case I am doing something wrong again. Using browseName instead of node id would be much more convenient.

closed 2021-07-20 by Girgitt