mikakaraila/node-red-contrib-opcua #330: Calling an OPC method with datatype 'NodeId' as input argument fails

schaffner, 2021-07-20:
Hello everybody

Following situation:
On my OPC server there are nodes with the datetype 'FileType'  and are organised in directories and subdirectories. They behave according to the [OPC specification Part 5 Annex C](https://reference.opcfoundation.org/Core/docs/Part5/#C).
Now I wish to delete one file node contained in a subdirectory. According to the previous mentioned specs I need to call the 'Delete' method of the subdirectory containing this file node and pass the NodeId of the file node I want to delete. From this I create the following msg object an pass it to the OPC Method node in Node-Red.
```
{
    "_msgid":"4712a5bbc7197ad6",
    "injectTime":1626779252617,
    "objectId":"ns=2;s=ServerRoot/MyDirectory/MySubdirectory",
    "methodId":"ns=2;s=ServerRoot/MyDirectory/MySubdirectory.Delete",
    "inputArguments":[
        {
            "name":"ObjectToDelete",
            "dataType":"NodeId",
            "value":"ServerRoot/MyDirectory/MySubdirectory/MyFileNode"
        }
    ]
}
```

Now as mentioned in the title this will fail and result in the following output on the Node-Red console window:

```
node:4656) UnhandledPromiseRejectionWarning: Error: Invalid variant arrayType: Scalar  dataType: NodeId value:ns=2;s=ServerRoot/MyDirectory/MySubdirectory/MyFileNode (javascript type = string )
    at constructHook (C:\ProgramData\npmglobal\node_modules\node-red-contrib-opcua\node_modules\node-opcua-variant\dist\variant.js:372:19)
    at new Variant (C:\ProgramData\npmglobal\node_modules\node-red-contrib-opcua\node_modules\node-opcua-variant\dist\variant.js:59:26)
    at BasicTypeSchema._coerceVariant [as coerce] (C:\ProgramData\npmglobal\node_modules\node-red-contrib-opcua\node_modules\node-opcua-variant\dist\variant.js:46:66)
    at initialize_value (C:\ProgramData\npmglobal\node_modules\node-red-contrib-opcua\node_modules\node-opcua-factory\dist\factories_schema_helpers.js:81:20)
    at Object.initialize_field_array (C:\ProgramData\npmglobal\node_modules\node-red-contrib-opcua\node_modules\node-opcua-factory\dist\factories_schema_helpers.js:108:17)
    at new CallMethodRequest (C:\ProgramData\npmglobal\node_modules\node-red-contrib-opcua\node_modules\node-opcua-types\dist\_generated_opcua_types.js:4174:52)
    at OPCUAMethodNode.node.callMethod (C:\ProgramData\npmglobal\node_modules\node-red-contrib-opcua\opcua\107-opcuamethod.js:206:35)
    at OPCUAMethodNode._inputCallback (C:\ProgramData\npmglobal\node_modules\node-red-contrib-opcua\opcua\107-opcuamethod.js:198:14)
    at C:\ProgramData\npmglobal\node_modules\node-red\node_modules\@node-red\runtime\lib\nodes\Node.js:203:26
    at Object.trigger (C:\ProgramData\npmglobal\node_modules\node-red\node_modules\@node-red\util\lib\hooks.js:149:13)
(node:4656) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 3)
```
From what I can see, I am guessing the data type 'NodeId' was not expected as an input argument.

Unfortunatly now I dont know for certain whetever this is because of bad input by me, an error in node-red-contrib-opcua or an error in node-opcua or a combination of any of the mentionend.

Any info on this problem would be appreciated

mikakaraila, 2021-07-20:
Yes you're correct, nodeID is missing data type. I have to add it...

mikakaraila, 2021-07-21:
Test with new v0.2.232

mikakaraila, 2021-07-21:
I just added NodeId as parameter type selection list, v0.2.233

[image]

schaffner, 2021-07-27:
Thanks for the quick fix, it took a while but finally I was able to do some testing with v0.2.233.

Unfortunately when injecting a msg object containing the properties'objectId', 'methodId' and 'inputArguments' with '"dataType":"NodeId"', I still get the same error as in the original post.
But manually entering the data with the user interface works perfectly fine. It seems to me, that the datatype check is not done when passintg the arguments  by msg object.

Going from this I was able to make a quick fix by simply checking the datatype and replacing the content of `inputArgument.value` with `opcua.coerceNodId(value)` in the method `node.callMethod` located in the file '107-opcuamethod.js' on line 209.  

While this works for me, I am sure there is a more elegant way of fixing it. 

Here is my fix, commented with `//jsc`
```
node.callMethod = async function (msg) {
      if (msg.methodId && msg.inputArguments) {
        verbose_log("Calling method: " + JSON.stringify(msg.methodId));
        verbose_log("InputArguments: " + JSON.stringify(msg.inputArguments));
//jsc - begin
	//Quick fix to coerce NodeId when input arguments are injected
	try{
	  msg.inputArguments.forEach((arg) => {
	    if(arg.dataType === "NodeId"){
	      arg.value = opcua.coerceNodeId(arg.value);
            }
          });
        }
	catch(err){
          node.error("Invalid NodeId: " + err);
	  return;
	}
//jsc - end
        const callMethodRequest = new opcua.CallMethodRequest({
          objectId: opcua.coerceNodeId(msg.objectId),
          methodId: opcua.coerceNodeId(msg.methodId),
          inputArguments: msg.inputArguments,
          outputArguments: msg.outputArguments
        });
.....
```

Do you see any problems with this fix? Thanks in advance.

mikakaraila, 2021-07-27:
Ack fix is needed as you cannot call opcua.coerceNodeId for the value. I will make it to my code and publish (today/tomorrow).
I was first fixing just UI based arguments.

mikakaraila, 2021-07-27:
I published v0.2.234 it contains this fix.

closed 2021-07-27 by mikakaraila