mikakaraila/node-red-contrib-opcua #331: Client refused to connect with Prosys OPC UA Simulation Server

faizal-a, 2021-07-23:
Current setup:

-> Prosys OPC UA Simulation Server running on my Windows 10 machine (same network)
-> Node-Red running on Raspberry Pi Debian(same network)
   - Node-Red v0.20.7
   - Node.js v14.17.1
   - opcua v0.2.233

I'm struggling to find a solution since the Client keeps reconnect without any luck.

I've tried both methods by changing the endpoint to server IP address(opc.tcp://192.168.X.XXX:53530/OPCUA/SimulationServer and also using hostname but remains the same.

OPC UA Server Endpoint has been set as follow;

[image]

Certificate on the Client node is None;

[image]

And here is the snapshot that I'm using for testing.

[image]

Kindly help me to fix this connection issue. Thanks in advance!

mikakaraila, 2021-07-23:
I have been using Prosys simulation server in my tests. So it must be something in environment or OS. Is there anything on console?

faizal-a, 2021-07-24:
It's worked now after I managed to open the connection of the port that hosts the server. Thanks for your help!

mikakaraila, 2021-07-24:
Please close

faizal-a, 2021-07-24:
Solved

closed 2021-07-24 by faizal-a