mikakaraila/node-red-contrib-opcua #332: Server creates wrong Namespace ID when using custom XLM Nodeset

PascalW123, 2021-08-04:
Hello,

im using the Server Node to create a OPC UA Server using a Custom .xml File (See the .txt File in attachments). 
[myserver.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/6932076/myserver.txt)

However when  the server is created instead of it being NS=1 for the nodes like specified in the .xml file it created all the objects and variables with the node ID NS=5. See Screenshot in attachments. This is a big issue because it means that all other nodes trying to connect to the specific variables will fail. Can you help me with this? It only happens on my laptop. When doing this on a different instance (i.E. a raspberry) it creates the server with the correct namespace index. 
[image]

mikakaraila, 2021-08-05:
Can you check node-opcua version?

PascalW123, 2021-08-05:
The Version is: 

node-opcua@0.7.4
and is
node-red-contrib-opcua 0.2.234

mikakaraila, 2021-08-05:
Update first node-opcua and test again

PascalW123, 2021-08-05:
Hello,

I updated/reinstalled both node-opcua and node-red-contrib-opcua. It still uses wrong NS. When I leave "Build default AddressSpace" checked. It creates the Default AddressSpace (VendorName Object and Variables) with the correct NS1 Namespace.

mikakaraila, 2021-08-05:
Node-red-contrib-opcua server supports registerNamespace

PascalW123, 2021-08-05:
I am not sure what you mean with that. How can I solve my issue with that?

mikakaraila, 2021-08-05:
No you can use JSON msg to server.
Traveling so I can provide example after back at home to access laptop.

mikakaraila, 2021-08-08:
Here is registerNamespace for server node:

[image]

PascalW123, 2021-08-11:
Sorry I am not sure how to use it. Even if I register new namespace. The OPCUA Server and its nodes will already be created with NS5 instead of proper NS1 that I want wont it? 
Could you show me example flow with the JSON inject?

mikakaraila, 2021-08-11:
Look OPCUA-TEST-NODES.json from examples.

[image]

PascalW123, 2021-08-11:
Ok. Thank you. This helped.

closed 2021-08-11 by PascalW123