mikakaraila/node-red-contrib-opcua #333: BadCertificateTimeInvalid

mrbrau, 2021-08-11:
Hello,

I'm using Node Red to simulate an opcua server. (Node-red-contrib-opcua V.0.2.232).
After one year i couldn't connect with a client do my server. UA Expert sends me the following error
[image]

I try to create a new compact server but i recive the same error. Can you tell me how to fix this error?

mikakaraila, 2021-08-11:
Certificate is not anymore valid, it is old now. Delete and generate new certificate (latest version will generate it if missing).

mrbrau, 2021-08-11:
I delete the following folder /home/pi/.config/NodeOPCUA-Default-nodejs
New Certificate was created but i recive the same error

mikakaraila, 2021-08-11:
Study more which files are actually under trusted folder. 
Server will copy client certificate file to trusted folder (or under cert).
It still uses old certificate if certificate ValidTo is same Date.

mikakaraila, 2021-08-16:
Code uses envPaths with node-red-opcua and it depends on OS you are running, look example below.

Example locations (with the default `nodejs` suffix):

		- macOS: `~/Library/Logs/MyApp-nodejs`
		- Windows: `%LOCALAPPDATA%\MyApp-nodejs\Log` (for example, `C:\Users\USERNAME\AppData\Local\MyApp-nodejs\Log`)
		- Linux: `~/.local/state/MyApp-nodejs` (or `$XDG_STATE_HOME/MyApp-nodejs`)
		-

FT2work, 2022-02-21:
Hi all,
I have the same error on some machine where the developed application is installed.
Can you specify all files and folders to be deleted to re-create clients certificates?
Thank you

closed 2023-07-04 by mikakaraila