mikakaraila/node-red-contrib-opcua #337: Error on reconnect

aschoenenberg, 2021-08-31:
I am running the opc ua client node with an unstable network connect. In the first few minutes my subscription is running perfectly. Sometimes it fails reconnecting an stuck in this state.

The logs show:
````
ERROR !!! callback has thrown en error  Error
    at Object.assert (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-assert/dist/index.js:13:21)
    at ClientSidePublishEngine._republish (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-client/dist/private/client_publish_engine.js:328:29)
    at ClientSidePublishEngine.__repairSubscription (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-client/dist/private/client_publish_engine.js:373:14)
    at repairSubscription (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-client/dist/private/client_publish_engine.js:180:18)
    at replenish (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:443:21)
    at /opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:448:13
    at eachOfLimit$1 (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:474:34)
    at awaitable (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:211:32)
    at eachOfGeneric (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:509:16)
    at Object.eachOf (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:623:16)
    at Object.awaitable (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:211:32)
    at ClientSidePublishEngine.republish (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-client/dist/private/client_publish_engine.js:182:15)
    at _ask_for_subscription_republish (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-client/dist/reconnection.js:110:32)
    at ask_for_subscription_republish (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-client/dist/reconnection.js:252:20)
    at /opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:3638:28
    at replenish (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:443:21)````
    
Is the client re-subscribing after reconnect?

It would be awesome if someone could look into this :) Thanks

mikakaraila, 2021-08-31:
Can you check node-opcua version? I´m not sure if @erossignon  can comment this one...

aschoenenberg, 2021-09-01:
Oh sorry, its version 0.2.236 (the latest I guess).

This could be the case, because the network isn't that stable :(

mikakaraila, 2021-09-02:
I asked node-opcua version, I expect it is v2.50.0

aschoenenberg, 2021-09-03:
Yes it is the v2.50.0.

I already change the switch to a gigabit one, so the network shouldn't be a problem anymore. There is still the same error

mikakaraila, 2021-09-05:
What will trigger error?
Can you find log that will show call stack that will show it?

aschoenenberg, 2021-09-06:
````
09:19:07.001Z :client_session_keepalive_manag:124    warning : ClientSessionKeepAliveManager#ping_server  Connection Break
09:19:07.300Z :client_session_keepalive_manag:124    warning : ClientSessionKeepAliveManager#ping_server  Connection Break
09:19:07.375Z :client_subscription_impl      :452   cannot find monitored item for clientHandle !: 1
09:19:07.618Z :client_session_keepalive_manag:124    warning : ClientSessionKeepAliveManager#ping_server  Connection Break
09:19:07.832Z :client_subscription_impl      :604   Receiving a notification for a unknown monitoredItem with clientHandle  1
09:19:07.889Z :client_session_keepalive_manag:124    warning : ClientSessionKeepAliveManager#ping_server  Connection Break
09:19:08.427Z :client_session_keepalive_manag:124    warning : ClientSessionKeepAliveManager#ping_server  Connection Break
09:19:08.580Z :client_subscription_impl      :452   cannot find monitored item for clientHandle !: 1
09:19:09.022Z :client_subscription_impl      :604   Receiving a notification for a unknown monitoredItem with clientHandle  1
09:19:09.026Z :client_subscription_impl      :452   cannot find monitored item for clientHandle !: 1
09:19:09.057Z :client_subscription_impl      :452   cannot find monitored item for clientHandle !: 1
09:19:09.238Z :client_subscription_impl      :604   Receiving a notification for a unknown monitoredItem with clientHandle  1
09:19:09.289Z :client_subscription_impl      :604   Receiving a notification for a unknown monitoredItem with clientHandle  1
09:19:10.497Z :client_subscription_impl      :604   Receiving a notification for a unknown monitoredItem with clientHandle  1
09:23:27.703Z :client_session_keepalive_manag:124    warning : ClientSessionKeepAliveManager#ping_server  Transaction has timed out ( timeout = 60000 ms)
09:23:33.619Z :client_session_keepalive_manag:124    warning : ClientSessionKeepAliveManager#ping_server  Transaction has timed out ( timeout = 60000 ms)
09:23:34.396Z :client_session_keepalive_manag:124    warning : ClientSessionKeepAliveManager#ping_server  Transaction has timed out ( timeout = 60000 ms)
09:23:34.971Z :client_session_keepalive_manag:124    warning : ClientSessionKeepAliveManager#ping_server  Transaction has timed out ( timeout = 60000 ms)
09:23:35.277Z :client_session_keepalive_manag:124    warning : ClientSessionKeepAliveManager#ping_server  Transaction has timed out ( timeout = 60000 ms)
09:23:52.571Z :client_session_keepalive_manag:124    warning : ClientSessionKeepAliveManager#ping_server  Connection Break
ERROR !!! callback has thrown en error  Error
    at Object.assert (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-assert/dist/index.js:13:21)
    at ClientSidePublishEngine._republish (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-client/dist/private/client_publish_engine.js:328:29)
    at ClientSidePublishEngine.__repairSubscription (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-client/dist/private/client_publish_engine.js:373:14)
    at repairSubscription (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-client/dist/private/client_publish_engine.js:180:18)
    at replenish (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:443:21)
    at /opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:448:13
    at eachOfLimit$1 (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:474:34)
    at awaitable (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:211:32)
    at eachOfGeneric (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:509:16)
    at Object.eachOf (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:623:16)
    at Object.awaitable (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:211:32)
    at ClientSidePublishEngine.republish (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-client/dist/private/client_publish_engine.js:182:15)
    at _ask_for_subscription_republish (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-client/dist/reconnection.js:110:32)
    at ask_for_subscription_republish (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-client/dist/reconnection.js:252:20)
    at /opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:3638:28
    at replenish (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:443:21)
ERROR !!! callback has thrown en error  Error
    at Object.assert (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-assert/dist/index.js:13:21)
    at ClientSidePublishEngine._republish (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-client/dist/private/client_publish_engine.js:328:29)
    at ClientSidePublishEngine.__repairSubscription (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-client/dist/private/client_publish_engine.js:373:14)
[... cut ...]
    at replenish (/opt/plcnext/.node-red/node_modules/node-red-contrib-opcua/node_modules/async/dist/async.js:443:21)
09:28:13.723Z :client_session_keepalive_manag:124    warning : ClientSessionKeepAliveManager#ping_server  Transaction has timed out ( timeout = 60000 ms)
09:28:37.827Z :client_session_keepalive_manag:124    warning : ClientSessionKeepAliveManager#ping_server  Connection Break
09:32:58.636Z :client_session_keepalive_manag:124    warning : ClientSessionKeepAliveManager#ping_server  Transaction has timed out ( timeout = 60000 ms)
````

Thanks for looking into it :)

mikakaraila, 2021-09-07:
Anything before client_session_keepalive_manager?
I wonder what is the first message as connection break happens?
It is not yet visible...

mikakaraila, 2021-10-05:
There seems to be something inside node-opcua: https://github.com/node-opcua/node-opcua/issues/1059

mikakaraila, 2022-03-06:
Closed as no more updates

closed 2022-03-06 by mikakaraila