mikakaraila/node-red-contrib-opcua #340: WRITE not working when originitang from http request

robbertvanhove, 2021-09-02:
When I trigger a WRITE (opc ua client) via an inject node, the value is properly written. But when I trigger this flow via a http request, it won't do the WRITE. 

In a first attempt, I removed the msg.res & msg.req properties in a function node. This way it worked! The only problem is that the http response cannot be send because it can't find the res object. 

[image]

mikakaraila, 2021-09-02:
OPC UA Client node was not designed to be used this way, sorry.
I think my code creates new msg object thus original input msg is lost.
But you could save it to context variable and retrieve it and merge after client sends msg to function node.

mikakaraila, 2021-09-19:
I have to study this a bit more. Original msg is used and it is send to output. Most probably some property overlap or JavaScript object issue.

grcamauer, 2023-09-08:
Was this ever solved?  I have a similar issue with calling an OPCUA method in response to a http request and not being able to return the result code.

mikakaraila, 2023-09-08:
Deep clone msg object as there are some fields most probably overlapping... and as you got response then "merge" msg with needed fields.

grcamauer, 2023-09-08:
Thank you for your quick reply.  Do you have any examples of this?Guillermo

mikakaraila, 2023-09-09:
I found one old example from somebody. Take a look and test this one:

[OPCUA-HTTP-MULTIPLE.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/12565904/OPCUA-HTTP-MULTIPLE.txt)

grcamauer, 2023-09-12:
Here is how I worked it out.  Thank you for your help Mika.

Guillermo

-- 
Guillermo Ruiz Camauer

[

    {

        "id": "d6b4c322.35a66",

        "type": "OpcUa-Method",

        "z": "4d24eae5.3b9b24",

        "endpoint": "7290d8b0.f713d8",

        "objectId": "ns=2;s=AdvancedMethods.OPCUA_Method_AdminBatch",

        "methodId": "ns=2;s=AdvancedMethods.OPCUA_Method_AdminBatch.AdminBatchOPCUAAdvancedControlHandler.ctc",

        "name": "CreateBatch",

        "inputArguments": [],

        "arg0name": "BatchName",

        "arg0type": "String",

        "arg0typeid": "",

        "arg0value": "",

        "arg1name": "BatchDescription",

        "arg1type": "String",

        "arg1typeid": "",

        "arg1value": "",

        "arg2name": "AdminCommandType",

        "arg2type": "String",

        "arg2typeid": "",

        "arg2value": "",

        "out0name": "ReturnCode",

        "out0type": "Int32",

        "out0typeid": "",

        "out0value": "",

        "x": 550,

        "y": 80,

        "wires": [

           [

                "58905587.9d5b0c",

                "465b052b.b4768c"
[... 391 lines cut ...]
    {

        "id": "7290d8b0.f713d8",

        "type": "OpcUa-Endpoint",

        "endpoint": "opc.tcp://10.96.123.14:4840",

        "secpol": "None",

        "secmode": "None",

        "none": false,

        "login": false,

        "usercert": false,

        "usercertificate": "",

        "userprivatekey": ""

    }

]

closed 2023-09-13 by mikakaraila