mikakaraila/node-red-contrib-opcua #341: Events and Alarms

lukinno1, 2021-09-03:
Hello

How can i create new events or alarms in my local server.?

I need create events and then i want read it.

Lukas

mikakaraila, 2021-09-03:
It's a bit more work... 
Do you need all different kind of alarms or just some specific one?

lukinno1, 2021-09-03:
I interested in about all alarms but it can be specific one like example.

mikakaraila, 2021-09-03:
I asked because Discrete alarm is simple, multi-state and level-alarm will need more parameters

lukinno1, 2021-09-03:
First show me discrete alarm.

mikakaraila, 2021-09-05:
Ok, I added new installDiscreteAlarm for the server commands. I need to test it and add some parameters so it will be usable.

mikakaraila, 2021-09-05:
First round done:
[image]

mikakaraila, 2021-09-05:
Inject and function node contains now needed extra parameters:
[image]
[image]

lukinno1, 2021-09-06:
[image]

mikakaraila, 2021-09-06:
I haven´t yet published this version. I will add LevelAlarm as it is almost as simple as DiscreteAlarm. Just few more parameters.

mikakaraila, 2021-09-06:
Limit alarm works, perhaps more testing, but simple as DiscreteAlarm:
[image]

lukinno1, 2021-09-07:
When will be publish this version?

mikakaraila, 2021-09-07:
Today, later after standard working hours.

mikakaraila, 2021-09-07:
Now it is available v0.2.238. I updated example OPCUA-TEST-NODES.json just remember to inject first focus AddVariable:

And after it DiscreteAlarm / LimitAlarm.

[image]

closed 2021-09-07 by mikakaraila