mikakaraila/node-red-contrib-opcua #342: Set port, which was read from file, for OPC UA server.

kihana, 2021-09-17:
Hello

Could you help me? Is it possible to read a PORT number from the file and next set it for the OPC UA server during the start of the flow?

mikakaraila, 2021-09-17:
You can read JSON file and inject it to server node. I can provide example later not today. No more programming today.

mikakaraila, 2021-09-18:
Here example:
[image]

Rename .txt files to .json and look reconnect tab.

[endpoint.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7189450/endpoint.txt)

[OPCUA-TEST-NODES.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7189451/OPCUA-TEST-NODES.txt)

kihana, 2021-09-20:
Thank you! Your solution works for OPC UA client node. Sorry I couldn't understand how can I do the same for OPC UA server node? And where I can find manual with description for such specific messages for OPC UA server and client nodes?

kihana, 2021-09-20:
And one more question. I see that on your flow OPC UA client node is active and also some debug information printed in debug sidebar. The question is which node printed this debug information OPC UA client node or some other node?

[image]

mikakaraila, 2021-09-20:
Just click node id on message then node-red will high-light corresponding node. It is node that you put red square on.

closed 2021-09-20 by mikakaraila