mikakaraila/node-red-contrib-opcua #344: Using OPC UA sign and encrypt

JGAD2021, 2021-09-29:
I am working with a siemens PLC. I want to use OPC UA to transfer data to node-red. I need secure OPC-UA communication. I made the configuration according to the siemens guidelines for the activation and configuration of the server including the generation of the certificates. Security policies based on Basic256Sha256 sign and encrypt are established. User authentication is also used. Client certificate authentication is activated. All setup is tested with UAExpert and works great, with and without security. When trying to connect with Node-red-opcua the communication without security works very well. But when activating security, I get uneasy with the handling of certificates. In the node-red configuration, the public certificate and the secure key in PEM format are requested. Can you give me more clarity on the handling of these certificates? Where should these certificates be? must they handle special permits? (I am using node-red on ubuntu 20). The version of node-red-opcua is 0.2.238. A secure OPC client configuration guide is available. I appreciate your guidance.

mikakaraila, 2021-09-30:
It is automated, node-opcua provides certificate manager that will provide client certificate through OPC UA to server and if server provides it´s certificate to client it will take care of storing it to folder.

Try to look folder trusted/cert, it depends on OS Windows has own location and Linux own. I haven´t check Linux location.

JGAD2021, 2021-09-30:
Thanks mikakaraila, working.

closed 2021-10-03 by mikakaraila