mikakaraila/node-red-contrib-opcua #345: OPCUaClient subscribe never recover when flow started before OPC server

ashok0617, 2021-10-02:
- Running flow on docker container.
- When flow starts, OPC server not yet ready and started.
- Subscription fails to register.
- Connection attempts continuously made.
- OPC server now running and available.
- OPCUaClient node connects
- Status shown keepAlive/listening
- When update node value on subscribed node, NO notification/subscription data being emitted: ISSUE
- After inject nodeid topic again then the subscription stored n started working normally as expected.
- This is also working OK if server connection made during flow starts and subscription succeeded.

Expected:
OPCUaCLient should be capable of recovering the nodeid subscription as soon as connection restored even subscription failed on flow start.

Workaround:
Create another flow for this node status and monitor the status change which is complex.
As per log - error "Invalid endpoints Error: " appears before to "create client". This is easily reproducible following step above. My subscription starts right after flow starts with delay of 1 second.
Let me know if I can help further on this.

ashok0617, 2021-10-03:
@mikakaraila 
I tested <Version: 0.2.238> and found it working but Anonymous user not connecting any more. Saw error in log as <**client node error on: error: [" opc ua connection error: cannot find anonymous user token policy in end point description"**]>. I did not see this issue so far. After I created an user under Administrator group on server and specified the user credential in OPCUa-Client node then connection succeeded.
I will recheck on test server, update here and if all good then close this issue. Seems I would need to upgrade the package to 0.2.238 but same time don't want server security update.

mikakaraila, 2021-10-03:
What is your server? It can be something on it´s initial and accepting subscriptions.

ashok0617, 2021-10-04:
> What is your server? It can be something on it´s initial and accepting subscriptions.

@mikakaraila , my opc server is kepware. I upgraded to <"node-red-contrib-opcua": "^0.2.238"> on another machine, ran through above steps - still able to reproduce. Please check log <OPCUA-CLINET-ERRORlog.txt> other ok logs also attached. I repeat the steps mentioned in my issue above.
[OPCUA-CLINET-ERRORlog.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7281492/OPCUA-CLINET-ERRORlog.txt)
[flow-start-opc-running-ok.txt](https://gith
[flow-start-opc-running-ok.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7281495/flow-start-opc-running-ok.txt)
ub.com/mikakaraila/node-red-contrib-opcua/files/7281493/flow-start-opc-running-ok.txt)

My subscription flow:
[image]

ashok0617, 2021-10-05:
> > What is your server? It can be something on it´s initial and accepting subscriptions.
> 
> @mikakaraila , my opc server is Kepware. I upgraded to <"node-red-contrib-opcua": "^0.2.238"> on another machine, ran through above steps - still able to reproduce. Please check log <OPCUA-CLINET-ERRORlog.txt> other ok logs also attached. I repeat the steps mentioned in my issue above. [OPCUA-CLINET-ERRORlog.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7281492/OPCUA-CLINET-ERRORlog.txt) [flow-start-opc-running-ok.txt](https://gith [flow-start-opc-running-ok.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7281495/flow-start-opc-running-ok.txt) ub.com/mikakaraila/node-red-contrib-opcua/files/7281493/flow-start-opc-running-ok.txt) [flow-start-connected-opc-stopped-started-ok.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7281583/flow-start-connected-opc-stopped-started-ok.txt)
> 
> My subscription flow: [image]
> 
> Node status turn to keepalive after OPC server started.
> 
> NOTE: The same version <0.2.238> tested on another computer(laptop) - windows 10 Home and found it working as reported above 2 days ago. There issue was that Anonymous user was NOT WORKING but after creating an user in OPC server, node was connecting and subscription worked perfectly including the case for which issue reported. Not sure why different behavior on different computer.
> 
> Conclusion
> 
> 1. Issue1 - Computer 1: Windows 10 Enterprise (OPC Server - Kepware 5.21 - same behavior on Kepware 6.10) - anonymous connection ok, subscription active when flow started and OPC server stopped then started. Subscription never work when OPC service stopped, flow started, OPC service started, node shows keepalive but changing value on OPC server never detected.
> 2. Issue2 - Computer 2: Windows 10 Home (OPC Server - Kepware 6.10) - Anonymous request does not work.
>    [image]
> 
> @mikakaraila , can you please giude me to set flag for verbose logging. Setting it in npm, as below, does not work.
> 
> > > npm run  --loglevel verbose

Log for anonymous user error
[image]

ashok0617, 2021-10-05:
@mikakaraila , now on debug I found that the pending message/subscription never being queued in a flow where issue being reproduced. So seems the <currentStatus> not reliable.
![image](https://user-images.githubusercontent.com/75435866/135951913-2b452889-dcdc-4241-bf98-be59b15b4c8b.png
[error.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7282398/error.txt)
[good.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7282453/good.txt)
)

Attached here error and good log.
Reason: the currentStatus (global to the module) changing in same timeframe.
[image]

Also attached the <102-opcuaclient.js.txt> with added logs. Remove txt for comparison. I appended as js not supported to upload.
[102-opcuaclient.js.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7282502/102-opcuaclient.js.txt)

mikakaraila, 2021-10-05:
You can get normal debug log by starting "node-red -v flow.json"

But ok now I know what to look thanks for the debugging.
Error occurs only if flow is started first and server later?

I expect error is here, can you check what is currentStatus when client is started?

```javascript
    if (!node.client || !node.session) {
        if (currentStatus == 'connecting') { // currentStatus is something else so msg are not put into the queue
          cmdQueue.push(msg);
```

ashok0617, 2021-10-05:
> You can get normal debug log by starting "node-red -v flow.json"
> 
> But ok now I know what to look thanks for the debugging. Error occurs only if flow is started first and server later?
> 
> I expect error is here, can you check what is currentStatus when client is started?
> 
> ```js
>     if (!node.client || !node.session) {
>         if (currentStatus == 'connecting') { // currentStatus is something else so msg are not put into the queue
>           cmdQueue.push(msg);
> ```

@mikakaraila , in further debug i found that processInputMsg is not called. I added a log in begining right after function call. I did not see that log - not sure why - further trying to add more logs.

<<
Error occurs only if flow is started first and server later?
>>
Correct. But one of small flow this is found working as stated in issue2 above. This is causing confusion why is it working for one flow but not other even creating the similar flow with delay inject of 1 second.

mikakaraila, 2021-10-05:
It is timing problem as session is created and subscription is done so fast server is not yet capable to create subscription.
I will make a change today, let´s see if it will help.

ashok0617, 2021-10-05:
> It is timing problem as session is created and subscription is done so fast server is not yet capable to create subscription. I will make a change today, let´s see if it will help.

thanks @mikakaraila - I have same opinion - must be timing issue.

mikakaraila, 2021-10-05:
Seems to old issue coming back, 4 years ago #58. Variable currentStatus contains new status messages / initialized as "".

Most probably caused this problem as actual session is not yet created.

mikakaraila, 2021-10-05:
Seems to work now as I start first node-red and after some time Prosys UA Simulation server, subscription that was injected at startup starts to show values.
I will publish v0.2.239

ashok0617, 2021-10-05:
> Seems to work now as I start first node-red and after some time Prosys UA Simulation server, subscription that was injected at startup starts to show values. I will publish v0.2.239

@mikakaraila , if we have new release I can check

mikakaraila, 2021-10-05:
Already published, sorry I didn´t send notice.

ashok0617, 2021-10-05:
> Already published, sorry I didn´t send notice.

Thanks for quick response @mikakaraila - I just applied the latest package but still issue happening. Looks message still never queued so after active session never expected message executed and hence node not going from active suscribing to subscribed state

[image]

ashok0617, 2021-10-06:
@mikakaraila pls fix as shown below. This is because of backoff firing and setting status to reconnect right after connecting https://github.com/mikakaraila/node-red-contrib-opcua/blob/65c0dbfe2417b8ba631524c52f771e4e39d8bdae/opcua/102-opcuaclient.js#L470

[image]

      if (!node.client || !node.session) {
        verbose_log("Not connected, current status:" + currentStatus);
        // Added statuses when msg must be put to queue
        const statuses = ['', 'create client', 'connecting', 'reconnect'];
        if (statuses.includes(currentStatus)) {
          cmdQueue.push(msg);
        } else {

I don't see edit option otherwise i would have generated PR

mikakaraila, 2021-10-06:
OK thanks, I will do after std. work at home. I will inform when I have published it.

ashok0617, 2021-10-06:
> OK thanks, I will do after std. work at home. I will inform when I have published it.

Pls let me know as son as done

closed 2021-10-06 by ashok0617