mikakaraila/node-red-contrib-opcua #349: opc-ua client connection problem

stesquality, 2021-10-31:
Hi.
I have an OPC-UA server in s7-1500 and I can read/write it through the UA-Expert client. On the other hand, I can also read/write through the node-red OPC-UA client on my computer. But in node-red installed raspberry pi shows reconnect attempt (yellow color below the OPC-UA client node). Could you suggest to me, how can I connect the server through the raspberry pi node-red OPC-UA client?
While Node Json v12.14.1 is installed in both computer and raspberry pi and v2.1.3 node-red is installed in raspberry pi.

mikakaraila, 2021-10-31:
Check first installation was successful then node-red console. It can be access rights...

stesquality, 2021-11-01:
Thanks for your kind reply, I am going to install a new rasbian and node-red. Is it nacssarry to install the 12.14 node. Jon version to use the OPC-UA node-red library or it will also work for the latest version

stesquality, 2021-11-04:
I have installed again raspbian, and now the OPC-UA server can connect but getting errors in the client (please see the attached pictures).  
[image]
[image]
[image]

mikakaraila, 2021-11-04:
I think client starts before server has initialized. Modify client and deploy only modified nodes to see if this is the case

stesquality, 2021-11-06:
What`s modification, I need to do in client, Also the program file is attached.
[Opc-UA server-client.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7490850/Opc-UA.server-client.txt)

mikakaraila, 2021-11-07:
Actually it takes more time to start server and server must have None or other endpoints active.

Client can connect, but SQL Query in the topic is not a nodeId.

Most probably you can continue your work with this one.
[OPCUA-Server-Client.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7493012/OPCUA-Server-Client.txt)

stesquality, 2021-11-08:
Thanks, it is working. Now, I want to move to my previous problem, where I need to connect the S7-1500 server with the Node-Red OPC-UA Client, while it is still showing, "Invalid EndPoint". While the UA-Expert connection to the server. The Node-Red file is also attached to this message. (I used that file around 6 months before and it was working but now getting connection error as mentioned above).
[OPC-UA Client.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7500232/OPC-UA.Client.txt)

mikakaraila, 2021-11-10:
Most probably some network / firewall problem if it works from the other computer.

mikakaraila, 2021-11-11:
And just in case check server log that it has still resources left for the client. In some cases it can be that there are not anymore sessions available on the server side and client doesn´t see or cannot give better error message at the moment.

stesquality, 2021-11-11:
Thanks for your kind reply, I tried ping PLC ip, and also telnet the IP with the port as shown in the attached pictures. in telnet I am only looking at a blank screen.
While, I got the OPC-UA log from the UA-Expert client as shown in the attached picture and also the log file attached.
[image]
[image]
[S7-1500 OPC-UA server.log](https://github.com/mikakaraila/node-red-contrib-opcua/files/7522967/S7-1500.OPC-UA.server.log)
[image]

mikakaraila, 2021-11-15:
Can you test with this environment variable set?

set NODEOPCUADEBUG=CLIENT{TRACE-REQUEST-RESPONSE-PERF-STAT}

It will print out to console how node-opcua handles messages on the protocol level.