mikakaraila/node-red-contrib-opcua #350: Browsing crash node-red

anxuae, 2021-11-02:
Hi @mikakaraila ,

I'm trying to do a browse of an OPCUA tree, but my flow is always crashing node-red.
Is there anything wrong with my flow?

Here is my flow:

```
[
    {
        "id": "edc89b61.2b3178",
        "type": "OpcUa-Client",
        "z": "2d2b75f0ff93f98c",
        "endpoint": "cf4490ec.250ef",
        "action": "browse",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": 10,
        "timeUnit": "s",
        "certificate": "n",
        "localfile": "",
        "localkeyfile": "",
        "securitymode": "None",
        "securitypolicy": "None",
        "name": "BROWSE",
        "x": 400,
        "y": 100,
        "wires": [
            [
                "5f41c17143d29bc5"
            ]
        ]
    },
    {
        "id": "f938ef3c.14866",
        "type": "inject",
        "z": "2d2b75f0ff93f98c",
        "name": "BinInRequest",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            },
            {
                "p": "collect",
                "v": "true",
                "vt": "bool"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "ns=3;s=\"DB_Comms_WCS\"",
        "payload": "",
        "payloadType": "str",
        "x": 190,
        "y": 100,
        "wires": [
            [
                "edc89b61.2b3178"
            ]
        ]
    },
    {
        "id": "5f41c17143d29bc5",
        "type": "debug",
        "z": "2d2b75f0ff93f98c",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 550,
        "y": 100,
        "wires": []
    },
    {
        "id": "cf4490ec.250ef",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://192.168.16.1:4840/Miniload_H20058_PLC/DataBlocksGlobal/DB_Comms_WCS",
        "secpol": "None",
        "secmode": "None",
        "login": false
    }
]
```

Here the error before node-red exits:

[image]

mikakaraila, 2021-11-02:
Can you take a screenshot how UaExpert will show address space on that folder/object?

I expect it could be something with the permissions... as error is coming inside node-opcua.
Error message: "a is not iterable" gives small hint to that direction.

anxuae, 2021-11-04:
Hi @mikakaraila 

Here is an example of the NodeId:

[image]

mikakaraila, 2021-11-04:
Does it contain something? I mean if you have root node and then folders with variables I cannot say yet anything.

Above should not be anything that will crash / break browse.

Like this:
[image]

anxuae, 2021-11-04:
Yes, there is a root folder. 

[image]

anxuae, 2021-11-04:
Yes there is a root folder

[image]

mikakaraila, 2021-11-05:
So next question at which node error happens, does the node have child nodes?

And what are child node access level / permission attributes?

anxuae, 2021-11-06:
Hi, I have no idea which node is causing the error. There are lots of nodes in the tree. I should browse it part by part. How can I select the node from which the browsing is done?

mikakaraila, 2021-11-07:
Just inject topic with the nodeId to client node with browse action:

[image]

anxuae, 2021-12-13:
Hello @mikakaraila, 

I've tried to reproduce the system without success. I will let you know if I have some news.
(you can close the issue, I will re-open it if necessary)

mikakaraila, 2021-12-13:
Ok you can re-open if you can repeat problem.

closed 2021-12-13 by mikakaraila