mikakaraila/node-red-contrib-opcua #351: Extension Object: Error when connecting two nodes at the output of the OPC Server

cedenoVF, 2021-11-03:
Hi,

I am using the READ MULTIPLE for reading several extension objects. However, after I tried to connect to nodes at the output of the OPC-UA server, and try to debug it, I get some error message.

1. This is the flow:

[image]

it doesn't matter how the functions are placed, I always get in the second branch the error below:

2. See below the output msg 2.

[image]

in the first branch I get always the information good and as expected from the PLC (btw, thanks a LOT for it, life it was easier now with this feature thumbs UP 👍  ):

[image]

3. Below you can find the error:

[image]

The "LEdgeBuffer_typeEdgeInterface" is one type of the objects that I am reading, though I don't understand the warning as I am getting that Extesion Objet in perfect shape in the debug **msg1** 

I am using the version of the contrib-opcua 2.243 

Thanks.

mikakaraila, 2021-11-03:
Hmm, for me it looks like node-opcua cannot solve something. I expect @erossignon should look and comment this one.

mikakaraila, 2021-11-07:
You can most probably ignore this error: 
[image]

As for me it looks like just a debug message from the node-opcua library.