mikakaraila/node-red-contrib-opcua #352: Use {{msg.topic}} in Item definition to Write

alesutto, 2021-11-03:
Hi,

I am trying to write a large amount of variables to a PLC for which the NodeIds are saved in global variables.
It seems like it is necessary to manually fill in the string with the NodeId in the Item section of the OpcUa-Item block.
I would like to know if it is possible to use for example {{msg.topic}} as input in the Item section instead of the specific string (using the necessary Item block) or if there is a way around.

[image]
The debug node shows the message inputted to the OpcUa-Item node.

Here the code.
[attachment (2).txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7467123/attachment.2.txt)

mikakaraila, 2021-11-03:
Item node is "fixed", but you can use standard function node to create needed items for the opcua.
Just set needed properties for the msg object like topic and datatype.

Use file node to read and send each line as msg to function node and then to client node.

I can provide example later if you don´t manage to do it today.

alesutto, 2021-11-03:
Solved!

Thanks for the quick answer :)

I tried again inputting the msg directly to the Client as 

{ payload : 50,
topic : "ns=3;s=\"Data_block_1\".\"input3\"",
datatype : "Int16"}

and it worked.

My issue was on the msg.topic, i previously used  

'"ns=3;s="Data_block_1"."input3""'' 

 and it wasn't recognized as correct

closed 2021-11-03 by alesutto

MrBoggi, 2023-11-13:
> Item node is "fixed", but you can use standard function node to create needed items for the opcua. Just set needed properties for the msg object like topic and datatype.
>
> Use file node to read and [...]

I'm trying to get this to work. Am I not alowed to use ns=4;i=7 to identify the variable to write to? 
[image]
As you can see, the topic is not being used. If I add the text I send in the topic as the item, it woks as expected.
ns=4;i=7 is the nodeID I'm trying to update.