mikakaraila/node-red-contrib-opcua #353: How to Create User.json File or use "SetUsers"-Command

PascalW123, 2021-11-15:
Hello,

im having issues with creating users for my OPCUA Server. 
I have created Json file like this: 
`{"username":"User1","password":"Password123", "role":"Administrator"}`

and I linked this in my OPCUA Server under Users File.
[image]
But when I try to connect I get this error:
[image]

I was not able to find an example of how to create and use users in your examples. 
Could you please help me out?

mikakaraila, 2021-11-15:
Start from the command line with the -v option. Then you can see how it loads users.json

C:\Users\karaimi\.node-red>node-red -v USERS2.json
15 Nov 13:25:43 - [info]

Welcome to Node-RED
===================

15 Nov 13:25:43 - [info] Node-RED version: v1.0.6
15 Nov 13:25:43 - [info] Node.js  version: v12.16.2
15 Nov 13:25:43 - [info] Windows_NT 10.0.19042 x64 LE
15 Nov 13:25:44 - [info] Loading palette nodes
15 Nov 13:25:46 - [info] Worldmap version 2.8.1
15 Nov 13:25:46 - [info] Dashboard version 2.27.0 started at /ui
15 Nov 13:25:46 - [info] Settings file  : C:\Users\karaimi\.node-red\settings.js
15 Nov 13:25:46 - [info] Context store  : 'default' [module=memory]
15 Nov 13:25:46 - [info] User directory : C:\Users\karaimi\.node-red
15 Nov 13:25:46 - [warn] Projects disabled : editorTheme.projects.enabled=false
15 Nov 13:25:46 - [info] Flows file     : C:\Users\karaimi\.node-red\USERS2.json
15 Nov 13:25:46 - [warn]

---------------------------------------------------------------------
Your flow credentials file is encrypted using a system-generated key.

If the system-generated key is lost for any reason, your credentials
file will not be recoverable, you will have to delete it and re-enter
your credentials.

You should set your own key using the 'credentialSecret' option in
your settings file. Node-RED will then re-encrypt your credentials
file using your chosen key the next time you deploy a change.
---------------------------------------------------------------------

15 Nov 13:25:46 - [info] Starting flows
Trying to load default users from file: c:/temp/users.json (current folder: C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\opcua)
15 Nov 13:25:46 - [info] [OpcUa-Server:4d4531ac.c402a] Trying to load default users from file: c:/temp/users.json (current folder: C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\opcua)
Loaded users: [{"username":"visitor","password":"qwerty","role":"Guest"},{"username":"JohnSmith","password":"1234","role":"Operator"},{"username":"Administrator","password":"HardPassword!","role":"ConfigureAdmin"}]
15 Nov 13:25:46 - [info] [OpcUa-Server:4d4531ac.c402a] Loaded users: [{"username":"visitor","password":"qwerty","role":"Guest"},{"username":"JohnSmith","password":"1234","role":"Operator"},{"username":"Administrator","password":"HardPassword!","role":"ConfigureAdmin"}]
: node set:C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\opcua\public\vendor\opc-foundation\xml\Opc.Ua.NodeSet2.xml,C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\opcua\public\vendor\opc-foundation\xml\Opc.Ua.AutoID.NodeSet2.xml,C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\opcua\public\vendor\opc-foundation\xml\Opc.ISA95.NodeSet2.xml
15 Nov 13:25:46 - [warn] [OpcUa-Server:4d4531ac.c402a] OpcUaServerNode: node set:C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\opcua\public\vendor\opc-foundation\xml\Opc.Ua.NodeSet2.xml,C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\opcua\public\vendor\opc-foundation\xml\Opc.Ua.AutoID.NodeSet2.xml,C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\opcua\public\vendor\opc-foundation\xml\Opc.ISA95.NodeSet2.xml
: create Server from XML ...
15 Nov 13:25:46 - [warn] [OpcUa-Server:4d4531ac.c402a] OpcUaServerNode: create Server from XML ...
15 Nov 13:25:46 - [info] Started flows
15 Nov 13:25:46 - [info] Server now running at http://127.0.0.1:1880/

mikakaraila, 2021-11-15:
Then with UaExpert:

[image]

PascalW123, 2021-11-15:
Hello,

thank you for your quick reply. 
However I have now run into new issue. 
While trying to fix this I updated the OPCUA Nodes Palette and now my .XML File will not work anymore.
[CSBME.zip](https://github.com/mikakaraila/node-red-contrib-opcua/files/7538491/CSBME.zip)
It worked fine before but now it never builds the namespace like it should. 
On my other PC with old Version it works with same file and looks like this: 
[image]
But on new one it it never shows the "CTX" Object and Variables under it.

mikakaraila, 2021-11-15:
Check this, it is not anymore one file, it is folder:

[image]

PascalW123, 2021-11-15:
Thank you!!! This fixed the Nodeset.
But users is not working yet sadly. 
I seeonly this when I enter OPCUsers.json
[image]

mikakaraila, 2021-11-15:
Again \ -> /

PascalW123, 2021-11-15:
Strange. Because it works on the Custom nodeset with \.
On the OPCUsers.json it doesnt work with /. See here: 
[image]

Since node-red shows me the error with \ as well and tells me to put it in here: "C:\Users\marvin\.node-red\node_modules\node-red-contrib-opcua\opcua" i tried that and only enter OPCUsers.json but sadly it doesn't seem to work

PascalW123, 2021-11-15:
It loads the Users but still not working in UaExpert 
[image]

mikakaraila, 2021-11-15:
UserIdentity [{"type":0}] is anonymous?

If I change user then I see this on UaExpert:

[image]

mikakaraila, 2021-11-15:
Do you have client node None/None options for Read/Write Multiple?

mikakaraila, 2021-11-15:
This way:

[image]

mikakaraila, 2021-11-15:
Sign means it will use user & password given below "Use credentials"

PascalW123, 2021-11-15:
Yes I do. type:0 is for anonymous I think yes. 
After I enabled all settings here (now I disabled Anonymous to test) it works. Even tho I choose none/none in UaExpert. 
[image]

I was able to connect with UaExpert using User and PW. Node, like write multiple can also connect using the User and PW. 
But only when setting SecurityPolicy and SecurityMode to "None". If I set them to Basic 256 and Sign I get this error:
[image]

mikakaraila, 2021-11-15:
Read comments above...

PascalW123, 2021-11-15:
Error when using Basic256 + Sign
[image]
Works when using None + None:
[image]

mikakaraila, 2021-11-15:
Depends on what you have active on server node allowed settings.

mikakaraila, 2023-02-28:
Not active anymore.

closed 2023-02-28 by mikakaraila