mikakaraila/node-red-contrib-opcua #355: Client Read Multiple Question

strike-back, 2021-11-20:
Hi there,
The contribution
https://github.com/mikakaraila/node-red-contrib-opcua/issues/198
describes writing this multiple times using the Node function.
Is there also the counterpart to reading multiple times?

Written multiple times
[image]

Read multiple times
[image]

mikakaraila, 2021-11-23:
In the examples folder should be OPCUA-TEST-NODES.json and in one tab is sub-flow that contains both cases.

strike-back, 2021-11-23:
Hi,

that's right, here I also write several times in one call.
But I get the data I want to read via the http-In node and send the result back to the http response node. Writing multiple times works great, but I would need a solution to read multiple times in this way.

Kind regards

mikakaraila, 2021-11-24:
Can you provide small example flow? It would help me to look what should be changed.

strike-back, 2021-12-07:
Hi,
Sorry for the late reply, was busy with work.
Here is the desired but not quite finished example to write several variables. In this way I would also like to be able to read multiple variables.

`[
    {
        "id": "b08e38fab6b7bc0d",
        "type": "tab",
        "label": "Flow 16",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "f6a5d319042b2d58",
        "type": "OpcUa-Client",
        "z": "b08e38fab6b7bc0d",
        "endpoint": "cc7a969e6bcbc504",
        "action": "writemultiple",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": 10,
        "timeUnit": "s",
        "certificate": "n",
        "localfile": "",
        "localkeyfile": "",
        "securitymode": "None",
        "securitypolicy": "None",
        "name": "OPC WRITE",
        "x": 710,
        "y": 340,
        "wires": [
            [
                "8300d55e757ff7f5",
                "390dafdc993be08f"
            ]
        ]
    },
    {
        "id": "0cb9acc12eac9c7d",
        "type": "http in",
        "z": "b08e38fab6b7bc0d",
        "name": "Http - Schreiben",
        "url": "/set-multiple",
        "method": "get",
        "upload": false,
        "swaggerDoc": "",
        "x": 140,
        "y": 340,
        "wires": [
            [
                "3d702561fb967b80",
                "f925f616407051e1"
            ]
        ]
    },
    {
        "id": "30b21ae77b6d2bed",
        "type": "http response",
        "z": "b08e38fab6b7bc0d",
        "name": "",
        "statusCode": "",
        "headers": {},
        "x": 1210,
        "y": 340,
        "wires": []
    },
    {
        "id": "390dafdc993be08f",
        "type": "function",
[... 176 lines cut ...]
        "name": "",
        "field": "payload",
        "fieldType": "msg",
        "format": "handlebars",
        "syntax": "mustache",
        "template": "<h1>Ergebnis Multiple-Write</h1>\n<table  class=\"table\">\n  </tr>\n  {{#payload}}\n  <tr>\n    <td>{{.}}</td> \n  </tr>\n{{/payload}}\n</table>\n\n\n\n",
        "output": "str",
        "x": 1040,
        "y": 340,
        "wires": [
            [
                "30b21ae77b6d2bed",
                "fc223765ed916cd5"
            ]
        ]
    },
    {
        "id": "cc7a969e6bcbc504",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://10.20.10.30:4840",
        "secpol": "None",
        "secmode": "None",
        "login": false
    }
]`

closed 2021-12-07 by strike-back

Eldarion85, 2022-02-14:
> Hi there, The contribution #198 describes writing this multiple times using the Node function. Is there also the counterpart to reading multiple times?
> 
> Written multiple times [image]
> 
> Read multiple times [image]

Why was the issue closed? I'd like to do exactly the same thing, but I can't get any further. The way from the example is of course possible. But that is not useful in practice. I would like to create a payload-array with the items (nodeId and datatype) to be read in a function, just like with WriteMultiple (of course without value). If I then pass the array into the client, I want an array with the values back.

Eldarion85, 2022-03-02:
Hello strike-back,

The multiple write works perfectly for me. You don't get a result when you write, except for an array, which variables could be written without errors (value: 0). You did it yourself in your screenshot above?!

Unfortunately the behavior at multiple read is completely different. Multiple read does not use arrays. You have to "register" the items individually at the node multiple read. Then a second message must be sent to the node that contains the string "readmultiple" in the topic. Now you get the value for all previously registered items. But unfortunately the answer comes for all items individually and not as an array. I also don't understand why I have to send a string with "readmultiple" to the node at all. It would make much more sense if this worked the same as write multiple. Just like you tried in the screenshot above. An array with the items as read multiple is sent and the result is an array with the current values. Of course you can reassemble all of this yourself and create an array from the returned messages. but then I can also work directly with read and don't need a read multiple.

Kind regards