mikakaraila/node-red-contrib-opcua #356: Allow Configuration via environment variables

binarynoise, 2021-11-23:
I am currently creating a subflow for easy replication of my flow.
Unfortunately I can't configure the item nor the server ip via environment variables.

How could it be possible to implement such feature? The item configuration should be quite simple, the server ip will be a bit more complicated as it's a global configuration node. For now, I put the opc-read and -write blocks out of the subflow, but it I can't also put there all the items.

mikakaraila, 2021-11-23:
Environment variables are not supported...

binarynoise, 2021-11-23:
I already found that out, but can we add support for it?

mikakaraila, 2021-11-23:
Environment variables can be used with some extra work but to make it "general" feature is quite a bi work.

Can you give examples where you would like to use them?

And I expect those must be generic not just like ${item1} etc.

binarynoise, 2021-11-23:
I got a flow with >100 Nodes, which gather data from a opc-ua-server, process them and display the data in a dashboard. From the dashboard, I can send data back to the server.
Now I need that flow 3 times and a subflow seemed the right candidate for me.

mikakaraila, 2021-11-23:
Easier & already working solution will be to read items from the file and inject them to item node.

binarynoise, 2021-11-23:
> I already found that out, but can we add support for it? Like with mqtt?

I found out, why it didn't work as I wanted, you can use the `${ENV_VAR}` -expression but you may not append or prepend it.
I went with relaying all the items through a function that edits the topic (item path) accordingly

mikakaraila, 2022-01-21:
You can use it inside function node this way:
https://nodered.org/docs/user-guide/environment-variables

[image]

closed 2022-01-21 by binarynoise