mikakaraila/node-red-contrib-opcua #357: Is there a way to do an historical access?

patufet99, 2021-11-28:
Hello,
as long as the server records the data, OPC-UA should allow historical access. Is it possible to retrieve historical data with the node-red-contrib-opcua implementation? I have been looking around, but I haven't found any example or information regarding how this could be achieved.
Thank you for your help.

mikakaraila, 2021-11-28:
Server node can do in memory historian. Just inject to server opcuaCommand installHistorian msg.

[image]

mikakaraila, 2021-11-28:
For the client I should add readHistory, I just noticed it is missing. It will be easy to add and also aggregates can be supported with some extra work like min, max & ave are available.

mikakaraila, 2021-11-28:
Aggregate support was not enabled on server, I added it and tested all available aggregates:

[image]

patufet99, 2021-11-28:
Whow! That's really cool!
Thank you so much!

patufet99, 2021-11-29:
I have defined a function:

```
msg.topic="ns=2;s=Device.Last fault.Nr";
msg.aggregate="raw";
msg.end = Date.now(); // GMT + 2h
msg.start = msg.end-10000; // Last 10 sec. history
return msg;
```
that I inject on the OPC-UA client with the HISTORICAL action.
That seems to work great!

For some reason the last element of the returned historical data array has always a "Null" dataType (as here below).

[image]

Additionally to the startTime and endTime, to get Historical access data you can usually define:
-a numValuesPerNode
-a boolean returnBounds
It is not clear for me how these can be injected in the flow.

Thanks again!

mikakaraila, 2021-11-29:
This was quick implementation, I will take a look and add those to msg as a parameter if node-opcua API allows to use them.

patufet99, 2021-11-29:
> Null comes perhaps as server initializes values, I will take look, but have you initialized variable before install historian?

If the history is empty two empty elements are returned.
If there is history data after the end time, the last element is not empty.
So, it looks like the bound elements are returned: element immediately previous and after the start and end time. If there is no such an element an empty element is returned.

mikakaraila, 2021-11-29:
Ok I already added these extra parameters.

I will make cleanup and some testing.

mikakaraila, 2021-11-29:
New v0.2.246 published, please test & feedback is welcome

patufet99, 2021-11-29:
I have added as indicated the new parameters
```
msg.topic="ns=2;s=Device.Last fault.Nr";
msg.aggregate="raw";
msg.end = Date.now(); // GMT + 2h
msg.start = msg.end-10000; // Last 10 sec. history
msg.numValuesPerNode = 5;
msg.returnBounds = false;
return msg;
```
They don't seem, however, to change much. 
 numValuesPerNode should limit the number of returned history data, but I continue getting all the data from msg.start to msg.end time.
As indicated previously if there is no history, an array with two empty elements is returned. My guess was that these were the bounds. By Setting returnBounds = false; I continue getting an array with two empty elements.

Best regards!

patufet99, 2021-11-29:
> They don't seem, however, to change much. numValuesPerNode should limit the number of returned history data, but I continue getting all the data from msg.start to msg.end time. As indicated [...]
The update seems to not have been done correctly, sorry.
Still testing ...

mikakaraila, 2021-11-29:
With my server I don´t get those null values. What is the server you are using?

patufet99, 2021-11-29:
> With my server I don´t get those null values. What is the server you are using?

It is a LabVIEW server. Checking with UaExpert, it seems to work OK. when there is no data it returns "No history values received from node ..."

mikakaraila, 2021-11-29:
Did you check with UaExpert History trend view numerical values?

Can you send screenshot? Status code is Good?

mikakaraila, 2021-11-29:
I expect that there is something with the server... what is OPC UA profile? Latest v1.04 or older, it could perhaps cause something.

patufet99, 2021-11-29:
You are right!

By looking at the detail of the records there are BadBoundNotFound!

[image]

mikakaraila, 2021-11-29:
Good, those can be omitted.

patufet99, 2021-11-29:
Yes, it looks like it is a problem of the Server implementation.
After the upgrade, it works!
```
msg.numValuesPerNode = 5;
msg.returnBounds = false;
```
With returnBounds=false; there are no array empty elements. If there is no history the array size is 0.
ValuesPerNode works as expected.

Thanks again for your invaluable help!

closed 2022-01-05 by mikakaraila