mikakaraila/node-red-contrib-opcua #358: Can read but can't write an r/w var defined with an alpha namespace, with error: BadNodeIdUnknown (0x80340000)

ppontre, 2021-11-29:
Hi Mika,
I'm using the latest version of the node, andi I found this strange behaviour. 
I can read regularly variables defines by strings, but but can't write them.
For example, a var defined like this:
ns=10;s=::AsGlobalPV:IPRRZP.batch
When I try to write it, I receive always this error:
BadNodeIdUnknown (0x80340000)
I checked that they're RW, and with UAExpert I can write them flawlessly.
If the variable is defined by numbers, all works welll.

Any Idea ?
Thanks in advance
Paolo

mikakaraila, 2021-11-29:
Check nodeId: ns=10;s=::AsGlobalPV:IPRRZP.batch with the UaExpert.

I remember that sometimes nodeId contains space and it can make it difficult to see it sometimes.
Or namespace index has changed.

ppontre, 2021-11-29:
Thanks Mika, 
We've just checked the nodeid  with UAExpert and it works well: we can read and write it.
Is there some way to get a deep log to see what happens in Nodered ?
Thx
Paolo

mikakaraila, 2021-11-29:
Another case can be that namespace index is with 2 digits. Most probably my code does not calculate it... I will check later today.

ppontre, 2021-11-29:
ok, i'll check the extra log option in the afternoon.....

mikakaraila, 2021-11-29:
Not needed, I found and fixed it already.

I will publish new version after some testing.

ppontre, 2021-11-29:
Great Mika !
I offer my scenario as a guinea pig, 🙂 if you tell me what to do.....

ppontre, 2021-11-29:
THX, I'll do it ASAP.....
Regards
Paolo

ppontre, 2021-11-29:
Thanks Mika, now it works OK !
GREAT work !
Regards
Paolo

closed 2021-11-29 by ppontre