mikakaraila/node-red-contrib-opcua #359: Is my OPCUA error caused by server certificate update?

leexuGit, 2021-11-30:
30 Nov 08:59:38 - [error] [OpcUa-Client:436d0525.ecd29c] Client node error on:  error: [" error at active reading: Transaction has timed out ( timeout = 60000 ms)"]
dispose when pendingTransactions is not empty 
00:59:38.366Z :client_session_keepalive_manag:145    warning : ClientSessionKeepAliveManager#ping_server  Session has been closed and should not be used to perform a transaction anymore
dispose when pendingTransactions is not empty 
dispose when pendingTransactions is not empty 
dispose when pendingTransactions is not empty 
(node:19122) UnhandledPromiseRejectionWarning: Error: Already disposing
    at OPCUACertificateManager.<anonymous> (/home/collect/collector/node_modules/node-opcua-pki/lib/pki/certificate_manager.ts:602:19)
    at Generator.next (<anonymous>)
    at /home/collect/collector/node_modules/node-opcua-pki/dist/pki/certificate_manager.js:31:71
    at new Promise (<anonymous>)
    at __awaiter (/home/collect/collector/node_modules/node-opcua-pki/dist/pki/certificate_manager.js:27:12)
    at OPCUACertificateManager.dispose (/home/collect/collector/node_modules/node-opcua-pki/dist/pki/certificate_manager.js:462:16)
    at OPCUACertificateManager.<anonymous> (/home/collect/collector/node_modules/node-opcua-certificate-manager/source/certificate_manager.ts:115:32)
    at Generator.next (<anonymous>)
    at /home/collect/collector/node_modules/node-opcua-certificate-manager/dist/certificate_manager.js:8:71
    at new Promise (<anonymous>)
(node:19122) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 51)
(node:19122) UnhandledPromiseRejectionWarning: Error: Already disposing
    at OPCUACertificateManager.<anonymous> (/home/collect/collector/node_modules/node-opcua-pki/lib/pki/certificate_manager.ts:602:19)
    at Generator.next (<anonymous>)
    at /home/collect/collector/node_modules/node-opcua-pki/dist/pki/certificate_manager.js:31:71
    at new Promise (<anonymous>)
    at __awaiter (/home/collect/collector/node_modules/node-opcua-pki/dist/pki/certificate_manager.js:27:12)
    at OPCUACertificateManager.dispose (/home/collect/collector/node_modules/node-opcua-pki/dist/pki/certificate_manager.js:462:16)
    at OPCUACertificateManager.<anonymous> (/home/collect/collector/node_modules/node-opcua-certificate-manager/source/certificate_manager.ts:115:32)
    at Generator.next (<anonymous>)
    at /home/collect/collector/node_modules/node-opcua-certificate-manager/dist/certificate_manager.js:8:71
    at new Promise (<anonymous>)
(node:19122) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 52)
(node:19122) UnhandledPromiseRejectionWarning: Error: Already disposing
    at OPCUACertificateManager.<anonymous> (/home/collect/collector/node_modules/node-opcua-pki/lib/pki/certificate_manager.ts:602:19)
    at Generator.next (<anonymous>)
    at /home/collect/collector/node_modules/node-opcua-pki/dist/pki/certificate_manager.js:31:71
    at new Promise (<anonymous>)
    at __awaiter (/home/collect/collector/node_modules/node-opcua-pki/dist/pki/certificate_manager.js:27:12)
    at OPCUACertificateManager.dispose (/home/collect/collector/node_modules/node-opcua-pki/dist/pki/certificate_manager.js:462:16)
    at OPCUACertificateManager.<anonymous> (/home/collect/collector/node_modules/node-opcua-certificate-manager/source/certificate_manager.ts:115:32)
    at Generator.next (<anonymous>)
    at /home/collect/collector/node_modules/node-opcua-certificate-manager/dist/certificate_manager.js:8:71
    at new Promise (<anonymous>)
(node:19122) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 53)
(node:19122) UnhandledPromiseRejectionWarning: Error: Already disposing
    at OPCUACertificateManager.<anonymous> (/home/collect/collector/node_modules/node-opcua-pki/lib/pki/certificate_manager.ts:602:19)
    at Generator.next (<anonymous>)
    at /home/collect/collector/node_modules/node-opcua-pki/dist/pki/certificate_manager.js:31:71
    at new Promise (<anonymous>)
    at __awaiter (/home/collect/collector/node_modules/node-opcua-pki/dist/pki/certificate_manager.js:27:12)
    at OPCUACertificateManager.dispose (/home/collect/collector/node_modules/node-opcua-pki/dist/pki/certificate_manager.js:462:16)
    at OPCUACertificateManager.<anonymous> (/home/collect/collector/node_modules/node-opcua-certificate-manager/source/certificate_manager.ts:115:32)
    at Generator.next (<anonymous>)
    at /home/collect/collector/node_modules/node-opcua-certificate-manager/dist/certificate_manager.js:8:71
    at new Promise (<anonymous>)
(node:19122) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 54)

30 Nov 08:59:43 - [error] [OpcUa-Client:11c13776.5651b9] Client node error on:  error: ["Cannot read endpoints: Error: Transaction has timed out ( timeout = 60000 ms)"]
Client node error on:  error: [" OPC UA connection error: Server end point are not known yet"]
30 Nov 08:59:43 - [error] [OpcUa-Client:11c13776.5651b9] Client node error on:  error: [" OPC UA connection error: Server end point are not known yet"]
00:59:45.880Z :client_base_impl              :498   the server certificate has changed,  we need to retrieve server certificate again:  Transaction has timed out ( timeout = 60000 ms)
ERROR !!! callback has thrown en error  TypeError [ERR_INVALID_ARG_TYPE]: The "data" argument must be of type string or an instance of Buffer, TypedArray, or DataView. Received undefined
    at Hash.update (internal/crypto/hash.js:82:11)
    at Object.makeSHA1Thumbprint (/home/collect/collector/node_modules/node-opcua-crypto/source/crypto_utils.ts:158:38)
    at /home/collect/collector/node_modules/node-opcua-client/source/private/client_base_impl.ts:502:63
    at /home/collect/collector/node_modules/node-opcua-client/source/private/client_base_impl.ts:620:21
    at /home/collect/collector/node_modules/node-opcua-client/node_modules/async/dist/async.js:2955:19
    at wrapper (/home/collect/collector/node_modules/node-opcua-client/node_modules/async/dist/async.js:268:20)
    at iterateeCallback (/home/collect/collector/node_modules/node-opcua-client/node_modules/async/dist/async.js:413:21)
    at /home/collect/collector/node_modules/node-opcua-client/node_modules/async/dist/async.js:321:20
    at /home/collect/collector/node_modules/node-opcua-client/node_modules/async/dist/async.js:2953:17
    at /home/collect/collector/node_modules/node-opcua-client/source/private/client_base_impl.ts:590:25 {
  code: 'ERR_INVALID_ARG_TYPE'
}
30 Nov 08:59:45 - [red] Uncaught Exception:
30 Nov 08:59:45 - Error: Callback was already called.
    at /home/collect/collector/node_modules/node-opcua-client/node_modules/async/dist/async.js:318:36
    at /home/collect/collector/node_modules/node-opcua-client/node_modules/async/dist/async.js:2953:17
    at /home/collect/collector/node_modules/node-opcua-client/source/private/client_base_impl.ts:581:36
    at /home/collect/collector/node_modules/node-opcua-secure-channel/source/client/client_secure_channel_layer.ts:1026:13
    at modified_callback (/home/collect/collector/node_modules/node-opcua-secure-channel/source/client/client_secure_channel_layer.ts:1328:17)
    at Timeout._onTimeout (/home/collect/collector/node_modules/node-opcua-secure-channel/source/client/client_secure_channel_layer.ts:1339:13)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)

mikakaraila, 2021-11-30:
Looks like you have some variable undefined:

The "data" argument must be of type string or an instance of Buffer, TypedArray, or DataView. **Received undefined**

leexuGit, 2021-12-01:
happen once in a while；
"node-opcua": "2.33.0"

mikakaraila, 2021-12-01:
I expect there was certificate manager introduced, it could have caused one time failure.

closed 2021-12-02 by mikakaraila

AutoTwisty, 2022-02-08:
Did you find a solution for this? I´m having a similar issue with the error "Already Disposing"