mikakaraila/node-red-contrib-opcua #360: Call Method broken in current release?

flex-altmraut, 2021-11-30:
Hi,

today I have updated to the latest version (from 0.2.220 to 0.2.246) and now Method nodes stops working with the following error.

I created a simple static Method node flow to reproduce:

[image]

[image]

```
(node:17) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'name' of null
    at /usr/src/node-red/node_modules/node-opcua-client-dynamic-extension-object/dist/get_extension_object_constructor.js:33:124
    at Generator.next (<anonymous>)
    at fulfilled (/usr/src/node-red/node_modules/node-opcua-client-dynamic-extension-object/dist/get_extension_object_constructor.js:5:58)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
(node:17) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 9)
```

Thanks

mikakaraila, 2021-11-30:
Sorry I was refactoring this to support ExtensionObject, I will finalize & fix this later today...

flex-altmraut, 2021-11-30:
Thanks for the quick reply, looking forward to the fix :)

mikakaraila, 2021-11-30:
I found other errors and improved output msg to contain also other results:

[image]

flex-altmraut, 2021-11-30:
Really appreciate your work. Thanks!

mikakaraila, 2021-11-30:
Method needs also outputArguments. Now you can define it and error handler will show error:

[image]

mikakaraila, 2021-11-30:
OK new version published v0.2.247

flex-altmraut, 2021-12-01:
Tested and it is working

closed 2021-12-01 by flex-altmraut