mikakaraila/node-red-contrib-opcua #361: Passing a json file with an array of topics to an OPC UA Client node

MaithiliDeshmukh, 2021-12-01:
Hi,
I am trying to pass a json file, containing an array of node IDs from a prosys server, to the OPC UA client node for reading the values. I want to be able to add or delete some of the nodes. Hence I have used a loop instead of fixed number of outputs. But I get an error saying msg.topic.indexOf is not a function. Can you please help me?
[something.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7634954/something.txt)
[flow1.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7634971/flow1.txt)
[image]

mikakaraila, 2021-12-01:
Greetings from Slush site event!

I simplified flow a lot:

[image]

mikakaraila, 2021-12-01:
Simple txt file with single line of nodeId with datatype:

nodeid.txt
ns=3;i=1001;datatype=Int32
ns=3;i=1003;datatype=Double
ns=3;i=1006;datatype=Double

MaithiliDeshmukh, 2021-12-02:
Wow..Thanks that is so much simplified! Also, I was not aware of msg.topic=readmultiple feature.

MaithiliDeshmukh, 2021-12-02:
Hi,

Thank you for simplifying this! I did not know about the msg.topic=readmultiple feature. Where can I read up more on this? Is there a database?

Thanks and Regards,

Maithili Deshmukh

[image]<https://user-images.githubusercontent.com/5184524/144295334-5c247330-b6ae-4ba4-9bee-ec5e5a67f0c2.png>

closed 2021-12-02 by MaithiliDeshmukh

mikakaraila, 2021-12-02:
Look examples folder, I have updated most of the features into the flow OPCUA-TEST-NODES.json

Each node has own Info/Description, look it as you open node properties.