mikakaraila/node-red-contrib-opcua #363: String Value read as Number

ppontre, 2021-12-02:
Hi Mika, 
I have another strange behaviour to submit:
I've got a  node with a string value, named as follows:
ns=10;s=::AsGlobalPV:IPRRZP.batch2
I can write string data without any problem, but I read back only a strange big number.
In UAExpert all works fine.
I attache some pics: Node definition, Result in both UAexpert and Node Red.
I hope they help.....

Regards
Paolo

[image]

[image]
[image]
[image]

mikakaraila, 2021-12-02:
What you inject inside READ?

ppontre, 2021-12-02:
I inject a timestamp.......
Ohhhhhhhh, You're right: I see its value !
But, Is it correct ?
I inject the same timestamp to many nodes and the other ones seem to work....

[image]
p

mikakaraila, 2021-12-02:
It was used earlier to set interval... or something should check from the code.

I normally inject empty string.

ppontre, 2021-12-03:
Thx Mika: 
I just re-create the node, checked and.... OK, it works fine (even with timestamp injected) !
Just an info for other users: that "msg" debug node is misleading: right place is AFTER the OPC-UA read node.
Regards
Paolo

closed 2021-12-03 by ppontre